hamishcoleman/thinkpad-ec #115: Unable to flash patched x230 v2.74

nkoester, 2019-02-17:
Hi, so I successfully created a patched.x230.img in which I only included the battery_validate patch based on g2uj30us.iso. Here the build info:

```
  $ git clone https://github.com/hamishcoleman/thinkpad-ec
  $ cd thinkpad-ec
  $ make patch_disable_keyboard clean
  $ make patch_enable_battery clean
  $ make patched.x230.img
  ./scripts/generate_deps Descriptions.txt >.d/generated.deps
  Downloading x230 BIOS 2.74 (G2ETB4WW) EC 1.14 (G2HT35WW)
  wget -nv -O g2uj30us.iso.orig https://download.lenovo.com/pccbbs/mobiles/g2uj30us.iso
  Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
  2019-02-17 18:37:01 URL:https://download.lenovo.com/pccbbs/mobiles/g2uj30us.iso [33927168/33927168] -> "g2uj30us.iso.orig" [1]
  scripts/checksum --mv_on_fail g2uj30us.iso.orig g2uj30us.iso
  touch g2uj30us.iso.orig
  ./scripts/ISO_copyFL2 from_iso g2uj30us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
  git submodule update --init --remote
  Submodule 'mec-tools' (https://github.com/eigenmatt/mec-tools) registered for path 'mec-tools'
  Cloning into '/home/nkoester/private/x230/software/thinkpad-ec/mec-tools'...
  Submodule path 'mec-tools': checked out '07a1b1407326488376afed6c596d37c53d7432b6'
  git submodule update
  make -C mec-tools
  make[1]: Entering directory '/home/nkoester/private/x230/software/thinkpad-ec/mec-tools'
  cc -Wall -O2 mec_csum_boot.c -o mec_csum_boot
  cc -Wall -O2 mec_csum_flasher.c -o mec_csum_flasher
  cc -Wall -O2 mec_csum_outer.c -o mec_csum_outer
  cc -Wall -O2 mec_encrypt.c -o mec_encrypt -lcrypto
  make[1]: Leaving directory '/home/nkoester/private/x230/software/thinkpad-ec/mec-tools'
  ./scripts/FL2_copyIMG from_fl2 x230.G2HT35WW.s01D3000.FL2.orig x230.G2HT35WW.img.enc.tmp
  IMG at offset 0x500000 size 0x30000 (FL2::prefix_ff x230.G2HT35WW.s01D3000.FL2.orig)
  mec-tools/mec_encrypt -d x230.G2HT35WW.img.enc.tmp x230.G2HT35WW.img.orig.tmp
  mec-tools/mec_csum_flasher -c x230.G2HT35WW.img.orig.tmp
  64c4 64c4 OK
  mec-tools/mec_csum_boot -c x230.G2HT35WW.img.orig.tmp
  b4b73a78 b4b73a78 OK
  f02cdcf8 f02cdcf8 OK
  3ff064dd 3ff064dd OK
  40da02e2 40da02e2 OK
  rm x230.G2HT35WW.img.enc.tmp
  mv x230.G2HT35WW.img.orig.tmp x230.G2HT35WW.img.orig
  cp --reflink=auto x230.G2HT35WW.img.orig x230.G2HT35WW.img
  ./scripts/hexpatch.pl --rm_on_fail --report x230.G2HT35WW.img.report x230.G2HT35WW.img x230.G2HT35WW.img.d/*.patch
  Attempting to patch x230.G2HT35WW.img
  Applying x230.G2HT35WW.img.d/006_battery_validate.patch -7294,7 +7294,7
  ./scripts/xx30.encrypt x230.G2HT35WW.img x230.G2HT35WW.img.enc.tmp
  + cd mec-tools
  + ./mec_repack ../x230.G2HT35WW.img ../x230.G2HT35WW.img.enc.tmp
  building ../x230.G2HT35WW.img.enc.tmp from ../x230.G2HT35WW.img
  b4b73a78 b4b73a78 OK
  f02cdcf8 f02cdcf8 OK
  3ff064dd 203e469e FIXED
  40da02e2 40da02e2 OK
  64c4 0962 FIXED
  98e8 7041 FIXED

  reverifying
  7041 7041 OK
  0962 0962 OK
  b4b73a78 b4b73a78 OK
  f02cdcf8 f02cdcf8 OK
  203e469e 203e469e OK
  40da02e2 40da02e2 OK
  cp --reflink=auto x230.G2HT35WW.s01D3000.FL2.orig x230.G2HT35WW.s01D3000.FL2.tmp
  ./scripts/FL2_copyIMG to_fl2 x230.G2HT35WW.s01D3000.FL2.tmp x230.G2HT35WW.img.enc.tmp
  IMG at offset 0x500000 size 0x30000 (FL2::prefix_ff x230.G2HT35WW.s01D3000.FL2.tmp)
  rm x230.G2HT35WW.img.enc.tmp
  mv x230.G2HT35WW.s01D3000.FL2.tmp x230.G2HT35WW.s01D3000.FL2
  echo "Buildinfo: v1-286-gd417ce-dirty (20190217) patched.x230.img" >x230.G2HT35WW.s01D3000.FL2.report
  echo "Built: `sha1sum x230.G2HT35WW.s01D3000.FL2`" >>x230.G2HT35WW.s01D3000.FL2.report
  echo "" >>x230.G2HT35WW.s01D3000.FL2.report
  cat x230.G2HT35WW.img.report >>x230.G2HT35WW.s01D3000.FL2.report
  sed -e "s%__DIR%`mdir -/ -b -i g2uj30us.iso.orig@@71680 |grep FL2 |head -1|cut -d/ -f3`%; s%__FL2%`mdir -/ -b -i g2uj30us.iso.orig@@71680 |grep FL2 |head -1|cut -d/ -f4`%" autoexec.bat.template >g2uj30us.iso.bat.tmp
  mv g2uj30us.iso.bat.tmp g2uj30us.iso.bat
  cp x230.G2HT35WW.s01D3000.FL2.report g2uj30us.iso.report
  echo "" >>g2uj30us.iso.report
  echo "Description: `scripts/describe g2uj30us.iso`" >>g2uj30us.iso.report
  cp --reflink=auto g2uj30us.iso.orig g2uj30us.iso.tmp
  cp --reflink=auto x230.G2HT35WW.s01D3000.FL2 x230.G2HT35WW.s01D3000.FL2.tmp
  cp --reflink=auto g2uj30us.iso.report g2uj30us.iso.report.tmp
  cp --reflink=auto g2uj30us.iso.bat g2uj30us.iso.bat.tmp
  touch -d @1 x230.G2HT35WW.s01D3000.FL2.tmp g2uj30us.iso.report.tmp g2uj30us.iso.bat.tmp
  # TODO - datestamp here could be the lastcommitdatestamp
  ./scripts/ISO_copyFL2 to_iso g2uj30us.iso.tmp x230.G2HT35WW.s01D3000.FL2.tmp 01D3000.FL2
  mcopy -t -m -o -i g2uj30us.iso.tmp@@71680 g2uj30us.iso.report.tmp ::report.txt
  mcopy -t -m -o -i g2uj30us.iso.tmp@@71680 g2uj30us.iso.bat.tmp ::AUTOEXEC.BAT
  mdel -i g2uj30us.iso.tmp@@71680 ::EFI/Boot/BootX64.efi
  rm x230.G2HT35WW.s01D3000.FL2.tmp g2uj30us.iso.report.tmp g2uj30us.iso.bat.tmp
  mv g2uj30us.iso.tmp g2uj30us.iso
  cp g2uj30us.iso patched.x230.iso
  cp g2uj30us.iso.report patched.x230.iso.report

  Your build has completed with the following details:

  Buildinfo: v1-286-gd417ce-dirty (20190217) patched.x230.img
  Built: c49c8d9b4b6fd14621f3e59cea04f113f9cee86e  x230.G2HT35WW.s01D3000.FL2

  Patches applied:
  x230.G2HT35WW.img.d/006_battery_validate.patch

  Description: x230 BIOS 2.74 (G2ETB4WW) EC 1.14 (G2HT35WW)
  ./scripts/geteltorito -o patched.x230.img.tmp patched.x230.iso
  Booting catalog starts at sector: 20
  Manufacturer of CD: NERO BURNING ROM
  Image architecture: x86
  Boot media type is: harddisk
  El Torito image starts at sector 27 and has 65536 sector(s) of 512 Bytes

  Image has been written to file "patched.x230.img.tmp".
  ./scripts/fix_mbr patched.x230.img.tmp
  mv patched.x230.img.tmp patched.x230.img

  Your build has completed with the following details:

  Buildinfo: v1-286-gd417ce-dirty (20190217) patched.x230.img
  Built: c49c8d9b4b6fd14621f3e59cea04f113f9cee86e  x230.G2HT35WW.s01D3000.FL2

  Patches applied:
  x230.G2HT35WW.img.d/006_battery_validate.patch

  Description: x230 BIOS 2.74 (G2ETB4WW) EC 1.14 (G2HT35WW)
```

After dd'ing it to a flash drive and booting from it I get the expected flash screen. On pressing a key the flashing goes through, ends with a long beep and success message. However, after the reboot the second screen / "flashing update" never appears. Any combination of UEFI and legacy booting does not change this behaviour. I am at a total loss here. What am I doing wrong? What can I do to make it work?

Current bios / ec:
```
UEFI BIOS Version G2ET94WW (2.54)
Embedded Controller Version G2HT33WW (1.12)
```

A similar issue I found could not help me with this: https://github.com/hamishcoleman/thinkpad-ec/issues/62

hamishcoleman, 2019-02-18:
First, you need to upgrade your BIOS and EC to the matching official Lenovo version - then you can install the patched version of the EC code.  

There may be something else going on that is stopping you from getting the second flashing screen, but it is very important that you dont mis-match the BIOS version and EC version as you might end up with a brick.

nkoester, 2019-02-18:
Oh wow, I did not see the notice in the first few sentences of the step-by-step instruction. So sorry for this - but I never managed to brick my device, so I got that going for me. Thanks for your help!

closed 2019-02-18 by nkoester