hamishcoleman/thinkpad-ec #119: Touchpad/Button issue

Vollblut, 2019-03-25:
Hello, 
i've flashed the BIOS on an X230 (2.73) successfully and everything works as intended (Back buttons at the arrow keys, keymapping, Volume buttons, you name it). 

i didn't get the trackpoint buttons (above the touchpad) to work though, Touchpad (also from an X220 with palmrest) works just fine including gestures. 

did i miss something?

Harinlen, 2019-04-05:
Having the same problems in W530

Vollblut, 2019-04-06:
I don't know if this pin was indeed for the trackpoint buttons, but after changing it to a new and correct isolation, it works like a charm.

Since there's no information on the web about it: it works perfectly with ISO keyboards as well.

Harinlen, 2019-04-08:
Which keyboard are you currently using? I figure out there is possibly several different keyboard model.

Vollblut, 2019-04-11:
dont know the FRU by heart, but it's the GER model (Germany). But since german keyboards seem to be rare, most of the time a "genuine" keyboard from ebay is a relabeled british or french keyboard. I assume it doesnt really matter which one you use and that it has nothing to do with the specific EC flash made by Hamish. 
But i just assume that, i'm really not a programmer, just an end-user who can edit configs and stuff.

Harinlen, 2019-04-12:
Thanks for your reply. I think the keyboard I bought should be a forged keyboard from a small factory. After I isolated those 3 pins, the light of the power button is not working. Will check after a new keyboard arrived.

Harinlen, 2019-04-24:
just replace a new US keyboard, apply the pin isolation. Two things are missing: the mid button of the trackpoint is not working and the power button light disappear. All the other thing works well. I remove the pin isolation, it works just fine. I have no idea why this happens. Maybe it has different pin layout?

dkogan, 2020-02-29:
Hi. I can't get the trackpoint buttons working at all on a t430 even though the rest works. Are you isolating the 2nd, 3rd and 4th pins as described in the wiki, or are you doing something else?

Harinlen, 2020-02-29:
I didn't isolate those pins, now sometimes it works, sometimes it needs a restart of OS. (W530)

dkogan, 2020-03-01:
So I figured it out. I was using electrical tape for the pin isolation, but this was too thick, and was interfering with the connectivity of the adjacent pins. Switching to a thinner isolating tape fixed everything. The trackpoint buttons work now. Scotch tape works. It's not intended to be used for this purpose, but hopefully it won't degrade over time.