hamishcoleman/thinkpad-ec #121: x230 - battery patch and keyboard patch conflict

Sobler, 2019-04-13:
I have been using x220 keyboard mod for a long time. Recently I installed new battery and applied battery mod. Battery is charging, but several keys changed behavior. For instance "Del" works as "Home", "Home" doesn't work. Then I used keyboard patch again and it repaired keyboard. But battery is not recognised as genuine and not charging. I repeated several times to be sure that it was not accidental.
My BIOS is G2ETB3WW (2.73). I used the ready-built images - in case if this fact can have any importance.

explit7, 2019-04-13:
Hi. I think,You need activate BOTH patches at the same time
Read the readme file how to do it.
What you make is - activating keyboard patch OR battery patch afterwards.
BUT you need BOTH

matrushko, 2019-11-05:
HI, 

Same to me. 
battery OK, in v2.76 , but keyboard keys changed *fn , supr, up, down, etc ...

what do you mean with ~at the same time~ , sorry i dont find the special characters in linux, i use windows normally.
how can i make them work toghether, step by step please, cause I am a windows user. 

thanks