hamishcoleman/thinkpad-ec #123: e330 battery whitelist problem

tarlabasi, 2019-04-22:
I have battery whitelist problem. 
I read support model information and I see e330 is same the l430 bios. but I didn't find l430 bios file. Can you help me?

hamishcoleman, 2019-07-31:
Unfortunately, the e330 (and l430) is using different hardware to the types of laptops that this repository can currently patch.  I doubt that the e330 and l430 are using the same BIOS files either.

Sorry.

closed 2019-07-31 by hamishcoleman

leecher1337, 2022-01-08:
Can you try the Battery hotpatch that I created for this machine, in case you have a battery to try?
Thanks: https://github.com/leecher1337/thinkpad-Lx30-ec

dartyukh-afk, 2026-03-31:
> [@dartyukh-afk](https://github.com/dartyukh-afk) Just as with every other patch in this repo:
> 
> ```
> git clone https://github.com/hamishcoleman/thinkpad-ec.git
> cd thinkpad-ec/
> make patch_disable_keyboard clean
> make patch_enable_battery clean
> make patched.e330.img
> ```

Yeah, already did this, thanks.