hamishcoleman/thinkpad-ec #125: x230 problem with x220 original batteries.

khseal, 2019-05-16:
The battery turns off for a short time. The battery charge drops to 1-5% and recovery charge.
I tested with multiple batteries. The battery turns off in any case. When working from the adapter or when running on battery. Just jumping the charge level while the laptop does not turn off.
With the battery from x230 there are no such problems.

sladen, 2019-05-16:
X230t battery can charge + discharge in X220t.

X220t battery can only discharge in X230t.

(Assuming I remembered the correct way around; if not, it's the opposite).

khseal, 2019-05-16:
I do not have tablet laptops.
I wrote about ordinary laptops after the patch.

hamishcoleman, 2019-08-06:
This sounds to me like the patch allowing "non genuine" batteries has not been applied.  The x220 did not check the battery, so you might have been using batteries that dont look genuine to the x230 (even if they are actually Lenovo batteries).

Can you confirm that the patch applied and all the steps looked like those described?

khseal, 2019-08-06:
The problem was gone, even though I didn't do anything to solve it.

closed 2019-08-06 by khseal