hamishcoleman/thinkpad-ec #126: Couldn't managed to have 29++ working on my x230

tberkayb, 2019-05-17:
I have done all the exact things in the read me file. (I changed the .OFF parts, just left the battery patch w/o OFF)

before patch in linux the battery info was shown (design power, brand etc.) but not charging
after the patch all battery info missing. The message is:
"~$ upower -i /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  power supply:         yes
  updated:              Sat 18 May 2019 01:06:43 EEST (97 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              0 Wh
    energy-empty:        0 Wh
    energy-full:         0 Wh
    energy-full-design:  0 Wh
    energy-rate:         0 W
    percentage:          0%
    capacity:            100%
    icon-name:          'battery-caution-charging-symbolic'
"

hamishcoleman, 2019-07-31:
The battery information should not have changed - this indicates some other problem.

The only thing that the patch changes is the result of the check for an original battery.  I dont know any reason why you would have started seeing this immediately after patching.

Have you tried reverting the patch?

tberkayb, 2019-07-31:
Thank you for your answer!

I tried it but it did nothing :(

-- 
*Tegin Berkay Budak*
Istanbul Technical University
Astronautical Engineering Undergraduate Student
Pars Rocketry Mechanics Team Member

*İletişim:*
E-mail: *teginberkaybudak@p <teginberkaybudak@gmail.com>arsroket.com
<http://arsroket.com>*
E-mail: teginberkaybudak@gmail.com <teginberkaybudak@hotmail.com>
Gsm: +905337238869

hamishcoleman, 2019-08-06:
Unfortunately, it sounds like something is actually wrong with your battery hardware.  I have not heard from anyone else about any issue like this with their patches - so I am assuming that it is just bad timing and unrelated to the patch

tberkayb, 2019-08-06:
hello there,

I think i was a little bit unlucky :/. Thank you for your contribution to
the thinkpad community!

Best Regards

Hamish Coleman <notifications@github.com>, 6 Ağu 2019 Sal, 18:48 tarihinde
şunu yazdı:

> Unfortunately, it sounds like something is actually wrong with your
> battery hardware. I have not heard from anyone else about any issue like
> this with their patches - so I am assuming that it is just bad timing and
> unrelated to the patch

-- 
*Tegin Berkay Budak*
Istanbul Technical University
Astronautical Engineering Undergraduate Student
Pars Rocketry Mechanics Team Member

*İletişim:*
E-mail: *teginberkaybudak@p <teginberkaybudak@gmail.com>arsroket.com
<http://arsroket.com>*
E-mail: teginberkaybudak@gmail.com <teginberkaybudak@hotmail.com>
Gsm: +905337238869

kmeaw, 2019-08-21:
I have reproduced the issue on my hardware.

kmeaw, 2019-09-09:
Bought another battery - now everything works fine.