hamishcoleman/thinkpad-ec #128: X230t Tablet BIOS 2.73 support? (GCETB3WW, 2019-04-18)

amit1z, 2019-06-16:
Hello, and thank you very much for your fight, and spreading theword! :)

Could you please also add support for:
* **X230t Tablet BIOS 2.73 of 2019-04-18** (GCETB3WW), EC 1.14 (GCHT25WW): [https://support.lenovo.com/us/en/downloads/ds029683](https://support.lenovo.com/us/en/downloads/ds029683)
in addition to the added
* x230 BIOS 2.75 of 2019-04-17 (G2ETB5WW), EC 1.14 (G2HT35WW): [https://support.lenovo.com/us/en/downloads/ds029187](https://support.lenovo.com/us/en/downloads/ds029187)

Much thanks in advance!

hamishcoleman, 2019-07-31:
I have just updated the X230t config to base off BIOS 2.73 (see e5b6325a470d6259b983630f863c23b484c81e0b)

(I also tried to add support for the newer X230t BIOS 2.74, but it seems that Lenovo have changed something in the image updates and it doesnt work)

Additionally, the EC firmware version has not changed since BIOS 2.58, so it is perfectly safe to use a patch built from an earlier BIOS version.

closed 2019-07-31 by hamishcoleman

amit4z, 2019-08-18:
Thanks so much!
X230t BIOS 2.74 also works now?

Is it possible to flash an older (-/+ ec patched) BIOS over a newer one, generally speaking? (Updating the README.md with this information would be great!)

hamishcoleman, 2019-08-18:
It is not a question about the BIOS version, but of the compatible EC firmware versions.  Since the BIOS 2.74 us using the EC 1.14, any patched image that also builds with a EC 1.14 can be used to flash your laptop.

Documenting this well is kind of hard - there is no hard-and-fast rule that I can make that is also easy for people to follow.

amit4z, 2019-08-25:
Got it! Thanks a bunch! I think you're already doing a great job!
Just do it as you feel comfortable.