hamishcoleman/thinkpad-ec #132: Cannot verify the SSL certificate of Lenovo driver download website

Africalifornia, 2019-08-01:
I cannot verify the SSL certificate of the download server.

After make _patched.x230.img_ I get below error.

```
Downloading x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)
ERROR: cannot verify download.lenovo.com's certificate, issued by ‘CN=GeoTrust RSA CA 2018,OU=www.digicert.com,O=DigiCert Inc,C=US’:
  Unable to locally verify the issuer's authority.
To connect to download.lenovo.com insecurely, use `--no-check-certificate'.
Makefile:164: recipe for target 'g2uj31us.iso.orig' failed
make: *** [g2uj31us.iso.orig] Error 5
```

I tried to update SSL certificates with
`sudo update-ca-certificates
`
Installed certificates manually from Digicert and GeoTrust to _/usr/share/ca-certificates/_   
`sudo dpkg-reconfigure ca-certificates`

But did not work.

Lenovo Support/Download Certificate is as below.

```
Common Name = support.lenovo.com

Subject Alternative Names = support.lenovo.com, b2r.motorola.in, commercial.lenovo.com, support.motorola.com, p.static.pub, supportapi.lenovo.com, smartsupport.lenovo.com, laapi.csw.lenovo.com, laapi.csw.lenovo.com.cn, mb2r.motorola.in, filedownload.lenovo.com, pos.motorola.in, customer360.lenovo.com, m.static.pub, mobilesupport.lenovo.com, test.download.lenovo.com, lenovomobilesupport.lenovo.com, cms.csw.lenovo.com, datacentersupport.lenovo.com, pcsupport.lenovo.com, cms.csw.lenovo.com.cn, motob2r.lenovouat.com, account.lenovo.com, us.download.lenovo.com, cn.download.lenovo.com, customer360.lenovo.com.cn, download.lenovo.com, pre-sales.lenovo.com

Issuer = GeoTrust RSA CA 2018
Serial Number = 0410194679FF351B16057121A44BFA93
SHA1 Thumbprint = 36430FC4310BD5F23B54A99870D4FB7024059835
Signature algorithm = SHA256 + RSA (excellent)
```
What is wrong here ?

hamishcoleman, 2019-08-06:
Unfortunately, this is an issue with your Operating System or your Network.  You have already done the update/reconfigure steps that would help.

Africalifornia, 2019-08-07:
> Unfortunately, this is an issue with your Operating System or your Network. You have already done the update/reconfigure steps that would help.

Dear Hamish,

I understood the problem source. I was trying to _make patched.x230.img_ from an **Ubuntu 18.04 OS LIVE** version. Actually, it was working on my USB and not installed on my device.
I think this OS version has missing certificates on its LIVE version, which prevents me to configure SSL certificates and that caused the installation problem from Lenovo drivers website.

I simply tried the same with **Mint 19.2 LIVE** and voila ! Even though it was running on USB, it worked. Here is the terminal log below;

```
mint@mint:~/thinkpad-ec$ make patched.x230.img
Downloading x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)
2019-08-07 00:34:24 URL:https://download.lenovo.com/pccbbs/mobiles/g2uj31us.iso [33927168/33927168] -> "g2uj31us.iso.orig" [1]
scripts/checksum --mv_on_fail g2uj31us.iso.orig g2uj31us.iso
./scripts/ISO_copyFL2 from_iso g2uj31us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
git submodule update --init --remote
Submodule 'mec-tools' (https://github.com/eigenmatt/mec-tools) registered for path 'mec-tools'
Cloning into '/home/mint/thinkpad-ec/mec-tools'...
Submodule path 'mec-tools': checked out '07a1b1407326488376afed6c596d37c53d7432b6'
git submodule update
make -C mec-tools
make[1]: Entering directory '/home/mint/thinkpad-ec/mec-tools'
cc -Wall -O2 mec_csum_boot.c -o mec_csum_boot 
cc -Wall -O2 mec_csum_flasher.c -o mec_csum_flasher 
cc -Wall -O2 mec_csum_outer.c -o mec_csum_outer 
cc -Wall -O2 mec_encrypt.c -o mec_encrypt -lcrypto
make[1]: Leaving directory '/home/mint/thinkpad-ec/mec-tools'
./scripts/FL2_copyIMG from_fl2 x230.G2HT35WW.s01D3000.FL2.orig x230.G2HT35WW.img.enc.tmp
IMG at offset 0x500000 size 0x30000 (FL2::prefix_ff x230.G2HT35WW.s01D3000.FL2.orig)
mec-tools/mec_encrypt -d x230.G2HT35WW.img.enc.tmp x230.G2HT35WW.img.orig.tmp
mec-tools/mec_csum_flasher -c x230.G2HT35WW.img.orig.tmp >/dev/null
mec-tools/mec_csum_boot -c x230.G2HT35WW.img.orig.tmp >/dev/null
./scripts/hexpatch.pl --rm_on_fail --report x230.G2HT35WW.img.report x230.G2HT35WW.img x230.G2HT35WW.img.d/001_keysym.patch x230.G2HT35WW.img.d/002_dead_keys.patch x230.G2HT35WW.img.d/003_keysym_replacements.patch x230.G2HT35WW.img.d/004_fn_keys.patch x230.G2HT35WW.img.d/005_fn_key_swap.patch
Attempting to patch x230.G2HT35WW.img
Applying x230.G2HT35WW.img.d/001_keysym.patch keysym
Applying x230.G2HT35WW.img.d/002_dead_keys.patch dead_keys
Applying x230.G2HT35WW.img.d/003_keysym_replacements.patch keysym_replacements
Applying x230.G2HT35WW.img.d/004_fn_keys.patch fn_keys
Applying x230.G2HT35WW.img.d/005_fn_key_swap.patch fn_key_swap
./scripts/xx30.encrypt x230.G2HT35WW.img x230.G2HT35WW.img.enc.tmp
+ cd mec-tools
+ ./mec_repack ../x230.G2HT35WW.img ../x230.G2HT35WW.img.enc.tmp
building ../x230.G2HT35WW.img.enc.tmp from ../x230.G2HT35WW.img
b4b73a78 b4b73a78 OK
f02cdcf8 f02cdcf8 OK
3ff064dd 3ff064dd OK
40da02e2 6d3996cc FIXED
64c4 7ac6 FIXED
98e8 a614 FIXED

reverifying
a614 a614 OK
7ac6 7ac6 OK
b4b73a78 b4b73a78 OK
f02cdcf8 f02cdcf8 OK
3ff064dd 3ff064dd OK
6d3996cc 6d3996cc OK
./scripts/FL2_copyIMG to_fl2 x230.G2HT35WW.s01D3000.FL2.tmp x230.G2HT35WW.img.enc.tmp
IMG at offset 0x500000 size 0x30000 (FL2::prefix_ff x230.G2HT35WW.s01D3000.FL2.tmp)
./scripts/ISO_copyFL2 to_iso g2uj31us.iso.tmp x230.G2HT35WW.s01D3000.FL2.tmp 01D3000.FL2
mcopy -t -m -o -i g2uj31us.iso.tmp@@71680 g2uj31us.iso.report.tmp ::report.txt
mcopy -t -m -o -i g2uj31us.iso.tmp@@71680 g2uj31us.iso.bat.tmp ::AUTOEXEC.BAT
mdel -i g2uj31us.iso.tmp@@71680 ::EFI/Boot/BootX64.efi
cp g2uj31us.iso patched.x230.iso
cp g2uj31us.iso.report patched.x230.iso.report

Your build has completed with the following details:

Built ISO: 4bc5feb73eac31c4da9f75c3861fe0085ffce8e8  patched.x230.iso
Based on code from: x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)
Buildinfo: v1-316-gb32edc (20190807) patched.x230.img
Built FL2: 5b0c4d2fc4a1521fb79d4288913a4a9af193f761  x230.G2HT35WW.s01D3000.FL2

Patches applied:
x230.G2HT35WW.img.d/001_keysym.patch x230.G2HT35WW.img.d/002_dead_keys.patch
x230.G2HT35WW.img.d/003_keysym_replacements.patch
x230.G2HT35WW.img.d/004_fn_keys.patch x230.G2HT35WW.img.d/005_fn_key_swap.patch

./scripts/geteltorito -o patched.x230.img.tmp patched.x230.iso
Booting catalog starts at sector: 20 
Manufacturer of CD: NERO BURNING ROM
Image architecture: x86
Boot media type is: harddisk
El Torito image starts at sector 27 and has 65536 sector(s) of 512 Bytes

Image has been written to file "patched.x230.img.tmp".
./scripts/fix_mbr patched.x230.img.tmp
INFO: Original Lenovo ISO contains a zero in MBR bootcode - attempting fix

Your build has completed with the following details:

Built ISO: 4bc5feb73eac31c4da9f75c3861fe0085ffce8e8  patched.x230.iso
Based on code from: x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)
Buildinfo: v1-316-gb32edc (20190807) patched.x230.img
Built FL2: 5b0c4d2fc4a1521fb79d4288913a4a9af193f761  x230.G2HT35WW.s01D3000.FL2

Patches applied:
x230.G2HT35WW.img.d/001_keysym.patch x230.G2HT35WW.img.d/002_dead_keys.patch
x230.G2HT35WW.img.d/003_keysym_replacements.patch
x230.G2HT35WW.img.d/004_fn_keys.patch x230.G2HT35WW.img.d/005_fn_key_swap.patch
```

From this log, it seems that _006_battery_validate.patch_ is not applied. And It is not set as _.OFF_

What can be the problem?

hamishcoleman, 2019-08-07:
It looks like you did not use the configuration commands.

have you run
```
make patch_enable_battery clean
```

(The old method of renaming the files is no longer used)

closed 2019-11-29 by hamishcoleman