hamishcoleman/thinkpad-ec #134: Verify that usb stick image is OK

ojn, 2019-08-04:
I submitted dd verification step using diff before:

https://github.com/hamishcoleman/thinkpad-ec/pull/131/files

Maybe there is some better method instead of using dd twice and diffing the images, what would that be?

hamishcoleman, 2019-08-06:
I had a thought about this and think that this is simpler:

```
cmp -n $(stat -c %s patched.x230.img) patched.x230.img /dev/sdx
```

It is one step and doesnt need any temporary files.  It might look more complex, but then I think that the "read this one number out of the half dozen output from dd" is pretty complex.

ojn, 2019-08-08:
That looks much faster, but it's completely silent when it finishes.
Maybe we could provide the exit status with echo and explain what it means

`cmp -n $(stat -c %s patched.x230.img) patched.x230.img /dev/sdx ; echo $?`

    0  images are identical
    1  images are different
    >1 in case of error

hamishcoleman, 2019-08-08:
or even

```
cmp -n $(stat -c %s patched.x230.img) patched.x230.img /dev/sdx && echo Image Verified OK
```

One thing I dont like is that this still requires the user to type the image name in twice.  This could be wrapped into a script, but then it doesnt serve as a simple documentation for showing how to do the check.

hamishcoleman, 2019-09-26:
I ended up using this just now:

`I=build/disk.img; cmp -n $(stat -c %s $I) $I /dev/sdb`

but it is still awkward