hamishcoleman/thinkpad-ec #135: make patched.w530.img downloads an older BIOS than is currently installed

Mamolll, 2019-08-05:
I currently have BIOS 2.75 installed on my W530. Running make patched.w530.img downloaded BIOS 2.72. I'm fairly sure the EC is the same in both versions (1.13), but I'm not sure if flashing the older patched version will cause issues. Should I downgrade to BIOS 2.72 before installing patched BIOS/EC? Or is there a way to download and patch the current 2.75 BIOS? Thanks to everyone who has help put this amazing project together btw!

hamishcoleman, 2019-08-05:
The EC firmware version has not changed (is it 1.13) between the older BIOS 2.72 and the BIOS 2.75, so you are good to use the patched.w530.img that you have built.

Unfortunately, Lenovo has changed something with the 2.75 release (From my investigation, it looks like they could have accidentally left out the EC firmware!), which has made it impossible to base the patch build on that version.

Mamolll, 2019-08-05:
Interesting. Just flashed and everything went smoothly. BIOS stayed at 2.75 and even my settings were retained! I was a bit uncertain about going the firmware route to remap the keys, but now I'm glad I did. Thanks again for this awesome tool!

hamishcoleman, 2019-08-06:
Fantastic!  Always good to hear success stories

closed 2019-08-06 by hamishcoleman