hamishcoleman/thinkpad-ec #136: Lost Space & Other Keys After Successful EC Flash on x230

qumage, 2019-08-16:
I have a Thinkpad x230, with a UK keyboard, running Ubuntu 18.04.3 LTS 64-bit. I have installed an x220 UK keyboard, and successfully flashed the EC. The following keys do not produce any response:

- F5
- F9
- F10
- Pause (this seems briefly to activate Ctrl-L)
- BackSpace
- Return
- Space

I can live without those function keys, but I would really like BackSpace, Return and Space to work.

I had problems until I removed the battery completely. I had to try to flash the EC three or four times before it worked. I got the 'Flashing EC' message, and then a strange noise for a second, and then several of the keys on my keyboard changed function. (That sounds like working, right?)

Sorry if I missed something obvious. I'm dreading you telling me it's the UK keyboard that's the issue. Thank you for creating this software! Any help would be appreciated. :)

hamishcoleman, 2019-08-16:
I've just had a look at the docs/table_matrix.txt file for the keys that you mention.  It looks like all the keys that you mention (except for the Pause key) are in the same column.

Did you test this keyboard before applying the patch?  If you applied the (recommended) insulating tape, you might have somehow affected that matrix row.

qumage, 2019-08-16:
It seems you may be right.

I opened up the keyboard and removed the electrical tape I applied. I tried
to clean the pins very lightly. When I booted, I did not regain the keys I
had lost. However, I now lost several new keys. I think I have lost Y, U, H
and J. These keys are also in the same column. I was not forceful, but I
guess I have either damaged the pins, damaged the pin contacts, obscured
them with adhesive, or torn the ribbon. I can see no tears, but if the
outer contacts are very vulnerable, then I could believe I have damaged
them. That, then, would be a dumb thing I would have done. I don't really
want to buy a new keyboard to fix this problem.

qumage, 2019-08-23:
I have just tested a new keyboard of the same model. All of my keys are now working normally, so I feel I can say with some certainty that I affected the matrix rows when I performed the electrical tape mod.

Thank you for holding this Issue open, and for your patience and advice.

closed 2019-08-23 by qumage