hamishcoleman/thinkpad-ec #137: provide prepatched .img files?

amit4z, 2019-08-18:
Would it be possible for you to provide prepatched .img files, or is this some IP/copyright-issue?

How about publishing sha256sums of such patched images for novices to quickly verify if everything went alright? (or to safely get the .img from somewhere else...)

hamishcoleman, 2019-08-18:
Unfortunately, this is a IP/copyright issue - The EC firmware clearly has its copyright owned by Lenovo, so I dont want to provoke anything by distributing it.

That being said, I should be able to update the CI system to generate hashes for all the targets and then make that easy for people to see (thus publishing the sha256sums, like you suggest)

hamishcoleman, 2019-08-18:
I cannot quickly think of an easy way to publish the data (and it only covers builds with default config options), but I have added something like this to in 87595382551514314025469a78db620a0b149cee. If you click through the github commits and the little green tick to get to the travis build log, you can see a report and some hashes as the last item in the build log (you will probably need to click the triangle twisty next to the words "make test.report")

amit4z, 2019-08-25:
I'm sorry I couldn't find it, but I'm still pretty inept at this github business...

But thanks a lot for taking up the suggestion. Would be so epic if that worked someday!!

amit4z, 2019-10-12:
Could you please paste me the sums for GCETB3WW for keymod with and without batterymod, I'm afraid I can't find it :( (maybe it's my browser)