hamishcoleman/thinkpad-ec #138: water-spill-resistance / drainage after keyboard-mod?

amit4z, 2019-08-25:
I was just finishing the mod on my **xx20-series keyboard** when I noticed the **2 small fluid-drainage gaps** in it's metal border-barrier (i.e. the bent up corners of the bottom plate) that goes almost all around and below the keys, and also [extends to the notches/nubs](https://www.thinkwiki.org/w/index.php?title=File:T410keyboardnubs.jpg) that would go under/into the X220(t)-palmrest:
* One gap is below the windows-key (left bottom's side) and
* Another between the left- and down-cursor-keys (right bottom's side).

My x230 Tablet's xx30-series **chiclet-keyboard** on the other hand has an about equally high plastic border all around and inbetween all keys, **no gaps**, and the notches of the plate are separately below, flat, and don't need to hold back water.
 And it seems despite my x230t having 2 drainage holes under the palmrest, and on the bottom of the case, the whole device would have to be turned on it's head in case of spillage?
 In fact, I tested it on the x230 keyboard alone (NOT RECOMMENDED! Now some of the keys are stuck!), and nothing got drained - it will eventually just follow gravity, and I suppose, would spill over the case and ports?

Among two of the great guides I followed, thinkpad-ec's recommended [thinkwiki.org](https://www.thinkwiki.org/w/index.php?title=Install_Classic_Keyboard_on_xx30_Series_ThinkPads) and the detailed [majek.sh post](https://majek.sh/en/classic-keyboard-for-lenovo-x230/), sadly none covered this. :( (Btw, this is another [amazing guide](https://imgur.com/a/fsRe6) you might want to link to, which clarifies the ribbon cable's connector sheets/flaps and the pins to isolate. As a first-timer it helped me a great deal!)

Has anyone ever tackled this problem or made an experience by mishap?

I could (should?) tape the x220-keyboard's drainage gaps shut (even though the x230t case has cutouts for them, at the inner edge of the palmrest, that it's own keyboard can't even utilize?).
 But if i have to sand down the notches **water might find a (wrong) way into the case?**
 Would the x220t-palmrest (+ apparently fits with x230t bezel?) help or make things worse?
 Apparently [it fits and doesn't?](https://www.reddit.com/r/thinkpad/comments/912t0m/x220t_palmrest_on_x230t/) (and [the X220 one?](https://www.reddit.com/r/thinkpad/comments/7bnnq6/question_x220_palmrest_on_the_x230/)). So, reports are conflicting. :(

Sorry for wall-o'-text - didn't know how to condense, or where to ask.

**Tl;dr:** Is the x230/t even made for water-drainage, or is this just a forgotten x220/t design remnant?

hamishcoleman, 2019-09-21:
Sorry, I cannot say that I have tested the spill resistance of the laptop!  I try to keep liquids away, regardless of the manufacturers claims.

amit4z, 2019-10-12:
Good call! I try to prevent such accidents too, but sometimes there might just be kids or pets around, or I'm on a train or elsewhere where I have little influence on others behaviour, or just be clumsy myself... It might just happen, and it'd give me peace of mind to know.
And from what I've read, for many people the spill resistance also seems to be a big selling point on thinkpads.

I couldn't find the X230 claim, but e.g. on Lenovo's ThinkPad X1 Carbon it reads:
> [_"we test all of our ThinkPads with 500ml of liquid"_](https://www.lenovo.com/us/en/landingpage/small-business/resources/download/thinkpad-x1-carbon-vs-late-night-latte/). (or this [ad with a T430](https://www.youtube.com/watch?v=8G7sZwobZWw)).

Here's a video showcasing the old keyboards **"Flat Tub"-spill mechanics with cut-outs** (which I tried to explain above) on a T410: [ThinkPad spill resistance - how does it really work](https://www.youtube.com/watch?v=mgppfpHqEWs?t=02m47s) (at [04:08](https://www.youtube.com/watch?v=mgppfpHqEWs?t=04m08s) claims X240 or later keyboards are sealed and need to be flipped around?)

The **closest answer** I could find is to [get and **glue the official water ducts with the X220 palmrest**](https://www.reddit.com/r/thinkpad/comments/bifo70/x220_keyboard_in_x230_waterproof/):
> _"You'll need to use palmrest cover from your original X220. X230 palmrest has different hole arrangement, so that spilled liquid will likely fry the motherboard or something. And, for some reason, most X220 palmrests sold on ebay or aliexpress come without water ducts attached, so, again, any spill will fry the motherboard, unless you glue these ducts to the purchased X220 palmrest... but then, you'll have to get the ducts somewhere, and the only place known to me is the original X220 palmrest. For some reason, Lenovo does not list the ducts as a separate FRU, so supposedly they should come with replacement palmrests, except that they don't (at least for those on aliexpress and ebay)."_

**If anyone ever looked further into this issue, or has more experience, please feel free to share it!**

One thing I've read a few times now from users reporting a spill (regarding **unmodded x230**'s): the **hardware usually seems to survive**, but like mine, the **keyboard most often is toast**:
X230: [_column of keys that don't work. I tried washing the keyboard with water to no success_](https://www.reddit.com/r/thinkpad/comments/5rp8pw/any_tips_on_cleaning_x230_keyboard_contacts/) > _"My end users are spilly executioner's of ThinkPads but when things leave a film I use a pencil eraser (just the soft one on the end of a pencil) and that usually cleans off the film and creates a bit of a better contact. If it doesn't...I replace the keyboard"_
X230: [_my non backlit one(from chicony too) looks perfect except that I killed it with water :/_](https://www.reddit.com/r/thinkpad/comments/bkwsb0/does_this_happen_to_anyone_else_basically_these/)
X230: [_I had to get a new keyboard._](https://www.reddit.com/r/thinkpad/comments/9emf9s/x230_joined_the_classic_7row_keyboard_club/)
X230: [_left it to dry upside down while I was in class, works flawlessly now._](https://www.reddit.com/r/thinkpad/comments/6x6xt6/x230_water_damage/)

**X220T**: [_keyboard is freaking out_](https://www.reddit.com/r/thinkpad/comments/5303wb/spilled_liquid_on_my_x220t_wont_boot/)

X200: [_20% of the keys isnt working after i accidentally spilled some water on it._](https://www.reddit.com/r/thinkpad/comments/bifo70/x220_keyboard_in_x230_waterproof/)

T440: [_keys are sticky, I’m hoping if I wear them out they’ll unstick but it’s rough_](https://www.reddit.com/r/thinkpad/comments/ad2n0k/guess_i_didnt_get_away_from_the_spill_unscathed/)
T440: [_spilt tea ... keyboard will need to be replaced ... trackpad is also fried_](https://www.reddit.com/r/thinkpad/comments/3dd3f2/possible_water_damage_keyboard_replacement/)
T460: [_keyboard has gotten kinda mushy in one corner_](https://www.reddit.com/r/thinkpad/comments/5n4ifo/how_to_clean_your_thinkpad/)