hamishcoleman/thinkpad-ec #141: T430s battery still not charging

mkecskes, 2019-09-08:
I have a T430s with a third party battery. I have applied the battery patch only. It seems the EC update went fine and "Flashing Embedded Controlled" was displayed after restarting. The battery warning is now gone, but it still isn't charging. I have popped the battery out and pushed it back, no avail. When entering BIOS it says the BIOS version is 2.63, which is odd, since the version I flashed is 2.75. The battery is in good condition: it was charged at around 60% and worked without AC power before discharging.

So my questions are:
* Was the update successful? It looks like it was, but it's not charging.
* Should I have updated the BIOS to version 2.75 before flashing the patched EC? That will be a bit tricky now because the updater from Lenovo requires a fully charged battery and this is the only battery I have and it's discharged.
* Can it be the issue that the battery is discharged? I tried with and without the battery in, still no success.

roby222, 2019-09-13:
same issue with a x230t

roby222, 2019-09-18:
@mkecskes after the step 5, but before the step 6 (building the image) you must launch 
` make patch_enable_battery clean`
after that you can continue with
step 6: `make patched.x230.img
`

without this command you don't patch the battery!

mkecskes, 2019-09-19:
@roby222 Yes, I knew and I have issued that command.

propiro, 2019-09-19:
Hello,
When i try to make patch_enable_battery clean i got 

make: "no rule to make patch_enable_battery". Stop

I want to flash my thinkpad so slice battery works.
Please advice.

hamishcoleman, 2019-09-21:
@mkecskes It does sound like the update has been successful - the most important clue is that the battery warning message has gone.  The EC version that goes with BIOS 2.63 is the same as the EC version from BIOS 2.75, so you do not need to worry about updating your BIOS.  The EC patch only updates the EC and does not touch the BIOS version - the image you booted was merely _based on files from_ the BIOS 2.75.

It might be that your battery is not working - after all, there was a reason that Lenovo put the fake battery check into their systems.

hamishcoleman, 2019-09-21:
@propiro You should open a new ticket with your issue - this ticket is about something else

mkecskes, 2019-10-05:
@hamishcoleman Indeed, it was the battery's fault. It works fine with another battery. Thank you!

closed 2019-10-05 by mkecskes