hamishcoleman/thinkpad-ec #143: Compatibility issues T430 bios 2.82

celroid, 2019-09-20:
If I roll back to 2.81 (just flash the older version) then will this work again?

hamishcoleman, 2019-09-21:
Given that this is a security enhancement, it would be strange of them to allow you to roll back to an earlier version.  That being said, the release notes https://download.lenovo.com/pccbbs/mobiles/g1uj49uc.txt have a section mentioning which versions can be rolled back, and this version is not mentioned.

In summary, please try it!  If you can successfully roll back to 2.81 then this will work again.

celroid, 2019-09-21:
I've successfully rolled back to 2.81 by just flashing it (had to disable secure rollback prevention before)
Then I've followed your guide and this video https://www.youtube.com/watch?v=MWZlktXFNJk
to flash a bios with the battery patch, and now my 27++ slice is charging.
Thank you.

hamishcoleman, 2019-09-21:
Thanks for that info - that is really handy to know.  I have added that to the compatibility warning in the readme.

gch1p, 2019-11-06:
Hey, I'm just want to say that it's possible to roll back from any version to any, their rollback prevention mechanism doesn't really prevent anything.

They have some sort of version check in the updating utilities, but all you need to do is just to call dosflash (or winflash) directly. In short, you need to edit the AUTOEXEC.BAT file in the bootable cd and replace the `command.com` line with `dosflash.exe /sd /file PATH\TO.FL1`, then write it to USB drive and boot in Legacy mode (not UEFI).

czubaka, 2019-12-23:
Hello celroid

I've just faced a problem with updating my t430s... its about step 6 in movie You linked... If this command
    make patched.t430s.img 

will always download the original file from Lenovo does it mean... it will be this locked newest bios ?
If yes so how can I change this step to make it work with 2.75 img file that I've downloaded manually ? (it's the latest unlocked bios for t430s) I will be grateful for help because I am kinda noob with linux :/

Srandista, 2020-02-05:
I can confirm, that indeed, you can downgrade from ECP 1.14 to 1.13 with simple BIOS downgrade. I went from 2.82 to 2.74 with official Lenovo BIOS flash utility, and then from 2.74 to 2.64 with [IVprep](https://github.com/n4ru/IVprep).

t430git, 2020-03-08:
> Given that this is a security enhancement, it would be strange of them to allow you to roll back to an earlier version. That being said, the release notes [...]

i have rolled back to bios v 2.81 but ec v1.14 ....should i proceed with

t430git, 2020-03-08:
> Hey, I'm just want to say that it's possible to roll back from any version to any, their rollback prevention mechanism doesn't really prevents anything.
> 
> They have some sort of version check in the updating utilities, but all you need to do is just to call dosflash (or winflash) directly. In short, you need to edit the AUTOEXEC.BAT file in the bootable cd and replace the `command.com` line with `dosflash.exe /sd /file PATH\TO.FL1`, then write it to USB drive and boot in Legacy mode (not UEFI).

can this method be used to roll back ec v1.14 ti ec v1.13...  by replacing fl1 with fl2 file

gch1p, 2020-03-08:
> can this method be used to roll back ec v1.14 ti ec v1.13... by replacing fl1 with fl2 file

No, don't replace anything. Take bootable ISO of some old version (like 2.60) and modify autoexec.bat as I said and downgrade BIOS. It will not downgrade EC, but then take that ISO again and use without modification. It then should reflash both BIOS and EC.

t430git, 2020-03-09:
> No, don't replace anything. Take bootable ISO of some old version (like 2.60) and modify autoexec.bat as I said and downgrade BIOS. It will not downgrade EC, but then take that ISO again and use [...]

thanks...

gch1p, 2020-03-09:
@t430git did it work?

t430git, 2020-03-10:
Actually we in india...celebrating holy festival...and my laptop is disassembled....to create bios backup externally...ill try on saturday once i recieve thermal paste ordered from aliexpress

t430git, 2020-03-22:
> 
> 
> @t430git did it work?

yes it did ,
i successfully rolled back to BIOS=2.81 and EC=1.13.
but windows installation updates it self and downloaded some lenovo power mng driver...
now battery shows pluged in.
at 77%.   ......     intermittent charging and discharging happens

t430git, 2020-03-22:
it must be conserving battery health and life....
so the Patch worked...
Great ..
Big thumbs UP.....guys

PR #161 (2020-03-11): README: adding downgrade info for laptops with locked version installed
Fixes:
#143 
#157 
#158