hamishcoleman/thinkpad-ec #145: Trackpad buttons not working

eugenevd, 2019-10-02:
Hi 

Since the trackpad buttons is part of the keyboard I'd think this question is relevant here.

I upgraded to the classic 7 row keyboard and flashed the EC as per your instructions with no errors.

My trackpad works, and the lower buttons work, but the trackpad buttons that is part of the keyboard (with the red lines) dont work - I assumed the EC updates will fix this along with the rest of they keys.

I've googled around, but haven't found any mention of this.

Any ideas?

hamishcoleman, 2019-10-03:
Did you perform the isolation step on the keyboard? (As seen in  http://www.thinkwiki.org/wiki/Install_Classic_Keyboard_on_xx30_Series_ThinkPads#Pinout) 

The signal lines for the trackpoint buttons are routed quite close to the signals that need to be isolated, so if the isolation is not done the melting caused by the short circuit can break the connection to the trackpoint buttons.

eugenevd, 2019-10-03:
I believe this is correct.
Could be that the right most is not completely exposed enough?

https://live.staticflickr.com/65535/48838541523_6a25290857_o.jpg

anounyym1, 2019-10-05:
Try put piece of paper between of that metal part and flex cable for better contact for trackpad buttons. That also fixes missing power LED. For best result you should use thin tape, like kapton.

eugenevd, 2019-10-07:
Ah, I assume you're telling me that the insulation tape I used could be slightly too thick and therefore raise the adjacent pins just enough so as to not make contact?
I'll retry with something thin and report back

eugenevd, 2019-10-07:
I didnt expect that to work, but indeed, covering the pins with a thinner piece of paper resulted in the 3 trackpad buttons of the keyboard working.

Thanks all

closed 2019-10-07 by eugenevd