hamishcoleman/thinkpad-ec #147: [x230] ThinkLight Stopped Working

ggg27, 2019-10-13:
###Basic Information

**Issue**: The ThinkLight no longer works after flashing on my x230 ThinkPad.
**Possible Temporary Solution**: Is is possible to revert to previous?

## More information

**Settings Used**: Default
**Device**: x230 ThinkPad
**Device used to Make Live USB**: Raspberry Pi 3 B+

hamishcoleman, 2019-10-16:
The only change that these patches may have applied to the thinklight is the keyboard combination used to activate it.

By default, these patches remap the keyboard layout - did you turn off that patchset or install an xx20 keyboard in your x230?

It is always possible to revert the software changes included in this patchset, just follow the instructions in the docs/CONFIG.md to disable both patchsets and build then install a new image.

ggg27, 2019-10-19:
Thanks, I figured the keyboard patch was not default.
_________________________________

Do I execute `make patch_enable_keyboard clean` after make `patched.x230.img`?

hamishcoleman, 2019-11-29:
First, do any configure options, then build the patched image to include those configured options