hamishcoleman/thinkpad-ec #148: X1 Carbon <> CVE-2019-6171

gunar, 2019-10-24:
I see the model Thinkpad X1 Carbon 5th is (not on the list of affected devices)[https://support.lenovo.com/gb/en/solutions/len-27764], still I'm getting the `thinkpad_ec: module verification failed: signature and/or required key missing - tainting kernel` error.

1. Do you think rolling back the kernel would solve it? What kernel version would you have me rollback to? Currently on 4.19.79-1.

2. Do you think we should add it to the table on README.md? 

Thank you,
Gunar

gunar, 2019-10-24:
Sorry I'm confused. This repo is about BIOS and I'm asking about Kernel. I didn't upgrade the BIOS and yet I'm having the above problem.

I understand my question is out of context for this project but any help would be greatly appreciated.

hamishcoleman, 2019-11-29:
This question is indeed out of scope for this repository.

The message you are getting is related to the Linux kernel loading unsigned modules - which you should take up with your distribution.

closed 2019-11-29 by hamishcoleman