hamishcoleman/thinkpad-ec #149: x230t patches 2.73 (GCETB3WW) instead of 2.74 (GCETB4WW) /+ offline use?

ami7az, 2019-10-31:
> _$ make patched.x230t.img_

tries downloading x230t BIOS 2.73 (GCETB3WW), gcuj32us.iso, instead of gcuj33us.iso (your last supported x230t BIOS, 2.74 (GCETB4WW) of 2019/06/25, which has newer security fixes and updated CPU microcode), that I put into the base of the _thinkpad-ec-master_ folder. (Putting the gcuj32us.iso there works)

[Descriptions.txt](https://github.com/hamishcoleman/thinkpad-ec/blob/master/Descriptions.txt) (and [README.md](https://github.com/hamishcoleman/thinkpad-ec/blob/master/README.md)) also list 2.74 as GCETB3WW, instead of GCETB4WW, if this could be a reason? Could you get it working? [#128](https://github.com/hamishcoleman/thinkpad-ec/issues/128) Or is there any way I need to force this, and could you please fix/update the scripts?

###
To run _thinkpad-ec_ offline, I just need the dependencies (_build-essential git mtools libssl-dev_) installed, _[mec-tools](https://github.com/eigenmatt/mec-tools/) with [openssl](https://www.openssl.org/)_ compiled and put in _thinkpad-ec-master/mec-tools/_, and the _supported BIOS iso_ file in the base folder, right?

hamishcoleman, 2019-11-02:
Firstly, the EC version is identical (EC 1.14) between the BIOS 2.73 and BIOS 2.74 - so there is no difference in the resulting EC patch.

Secondly, with BIOS 2.74, Lenovo appears to have released a broken BIOS update image - it does not contain a valid EC update blob - so we cannot base the build from that version.

For offline use, you are right - once the packages are installed and the mec-tools sub-module has been downloaded, all that is needed is the correct BIOS .iso file (named .iso.orig in this case)

ami7az, 2019-11-13:
Thank you!

Are these sha256sums correct for the patched x230t BIOS 2.73 (GCETB3WW)?

> 67f848bb5f380cbbf2d831b109fe06630dbcf0ccaa0c6e38f70693ba5e47cfbf  gcuj32us.iso.orig

> $ make patch_enable_keyboard clean && make patch_disable_battery clean && make patched.x230t.img
> **77021b5d48594eff54117e6b0f83bf1b7673f3c846466b6cacd3b88791d1d29d  x230t-2.73-KEYBOARD-patched.x230t.img**
> 59325c103c74148571cbc12e0509de0a037cd6b29a67f43a2b38348ecd05adcd  x230t-2.73-KEYBOARD-patched.x230t.iso

> $ make patch_enable_keyboard clean && make patch_enable_battery clean && make patched.x230t.img
> **297ebb699cce1a53727e5a5161f164e3ba24e0108cabc1b33b5d4a2c1e258b54  x230t-2.73-KEYBOARD+BATTERY-patched.x230t.img**
> 5ab92e136fd36f1943c0f8386e0d64466039a0d8122a62b518b739a0ade931fd  x230t-2.73-KEYBOARD+BATTERY-patched.x230t.iso

> $ make patch_disable_keyboard clean && make patch_enable_battery clean && make patched.x230t.img
> **c21169c94de4b95da6a2fac5118257f020053bbd6e2a26e40236449469665c78  x230t-2.73-BATTERY-patched.x230t.img**
> 6a5f869e5861fb5bbe57a052fa641ace86ded2b4e57d37fc209b85ab5f6e140a  x230t-2.73-BATTERY-patched.x230t.iso

Can the .iso's also be written with 'dd if=patched.x230.iso of=/dev/sdx bs=4M status=progress conv=fsync', or do I need to use the .img version?

Perhaps providing the sha256sums of all last supported versions in the README.md or adding a _thinkpad-ec-all-supported-patched.sha256_ would be helpful to new users, as they will be unlikely to change after CVE-2019-6171?

Can you disable secure rollback prevention on all models?

If you flash a newer original BIOS over the patched one, will it keep the EC patches, or overwrite them?

In case of the latter, it would mean I have to either switch to coreboot if I wanted to keep getting security fixes and CPU microcode updates, or just live with the missing and 'switched' xx20 keys / original batteries, right?

ami7az, 2019-12-15:
It worked!!
But I compiled again and noticed _**different sha256sum's**_. 1st time I think I compiled it with a required src file that I took from `openssl-1.1.1d.tar.gz` myself, this time I had `$ sudo apt-cache show openssl`  _Version: 1.1.1c-1 Depends: libc6 (>= 2.15), libssl1.1 (>= 1.1.1)_ installed (in case this matters). Both times with `mec-tools-master.zip` of 19-03-25 (latest) and on debian 10.
And AC and a charged battery had to be plugged or it wouldn't reboot for the actual flashing. (Even though it said `Flashing finished.` and `BIOS is updated successfully` beforehand.)

With BIOS Legacy boot either of these 2 works fine from [Grub](https://www.gnu.org/software/grub/) 2.02 and with [Syslinux' Memdisk](https://wiki.syslinux.org/wiki/index.php?title=MEMDISK) file in the same folder (here /iso/), so no need to dd to a stick:

> menuentry "patched.x230t.img" {
>    linux16 /iso/memdisk
>    initrd16 /iso/patched.x230t.img
>}
>
>menuentry "patched.x230t.iso" {
>    linux16 /iso/memdisk iso
>    initrd16 /iso/patched.x230t.iso
>}

(The original gcuj32us.iso on the other end seems to work neither by memdisk nor by flashing to USB)

It would be great if you could still verify these sums, and **if there were a way to always get reproducible builds?**

> $ make patch_enable_**keyboard** clean && make patch_disable_battery clean && make patched.x230t.img
19091630373cafd5703808f9e9a62eb1519578de13ddcf4f915a4c5de1e0fc13  patched.x230t.iso
e977af012c741e0c3572ff65b86827382e3c54118621c403b301b423b4790350  patched.x230t.img
> $ make patch_enable_**keyboard** clean **&**& make patch_enable_**battery** clean && make patched.x230t.img
7e068c1eec6e4477fd5bd19c3bbf25adee0830aaff87078c87670129b153916f  patched.x230t.img
e356ff00ca285d410452c4cbe31451a140684c20b4f765650be21ba0510a1205  patched.x230t.iso
> $ make patch_disable_keyboard clean && make patch_enable_**battery** clean && make patched.x230t.img
c2c4b99a9da283320e10934b57661a152a71c3dddfd1d09581ef3817eaafdd68  patched.x230t.img
a4ba70422bf4055833ff7e34d1dec73d8e8327dc76667796de1db02e9eed7cca  patched.x230t.iso

##
And I've noticed the update readme's like gcuj32us.txt (v2.73) often mention things like these:
>- If the UEFI BIOS has been updated to version **2.68 or higher, it is no longer**
>  **able to roll back to the version before 2.68** for security improvement.

So I suppose one of these days we can't roll back to a _thinkpad-ec_-supported version anymore. It would be amazing **if someone could figure out a way to directly modify or flash the EC** e.g. with improved MEC-tools by then, and this way we might even be able to do it under coreboot!

##
Thanks so much for your project, Hamish!! The old keyboard is by magnitudes better and the patching worked like a charm! The only thing I'm missing a little so far is the caps-lock light. (And sometimes the trackpoint keeps drifting, but this is most likely a keyboard thing.)

gch1p, 2020-01-10:
>  I suppose one of these days we can't roll back to a thinkpad-ec-supported version anymore

You can: https://github.com/gch1p/thinkpad-bios-software-flashing-guide