hamishcoleman/thinkpad-ec #150: X230 EC Reset - Backlit keyboard stop working

kasik96, 2019-11-26:
Hi, so i used classic keyboard on my X230 for some time, and i wanted to revert it and put back my X230 keyboard with backlight back. I put my keyboard back and refleshed EC with 
make patch_disable_keyboard clean
 And everything is working fine again, except my back-light on keyboard. Light from the top of screen works (Fn + Space) but only it. Also i was researching a little, but i dont think my fuse blow up, because when i press Fn + Space, i get off or on. People who fuse blow up is dooing the classic think that Fn + Space first time its back-light (so if blowed fuse its nothing) than screen light and then nothing. I got always on or of top light. 

I hope i maked it clear to understand. I think its problem with EC. Thanks for any help :)

mszumilak, 2019-11-27:
Try [Fn]+[Delete] then. If it works it means that you have installed image patched for classic 7-row keyboard (like in x220).

kasik96, 2019-11-27:
Thanks for answer but thats not the problem. I can turn on light with [Fn] + [Space] but only that small light above the screen (thinklight), but my keyboard backlight does not work. I cannot start it with any shortcut. Just thinklight works not keyboard backlight. And it worked before.

hamishcoleman, 2019-11-29:
Sorry, I dont have a lot of experience with putting the x230 keyboard back in.  However, it is clear that the EC can detect a non backlit keyboard and just skip that part of the backlight sequence. Since you isolated your x230 pins, there shouldnt be any blown fuses, but perhaps there is some other part of the backlight hardware that we dont understand.

kasik96, 2019-12-04:
Thanks for reply. It would be nice if there would be anyone with experience putting back back-lit keyboard and succeeded. But until then, i am gonna use Classic keyboard again. Thanks for help.

kasik96, 2019-12-07:
Problem solved, i am was thinking i had keyboard with backlight, but i installed wrong one apparently. So, i am sorry for this misunderstanding. EC flash works as it should. Thanks.

closed 2019-12-07 by kasik96

hamishcoleman, 2019-12-07:
Aw, snap!  I guess that's a good answer - Thanks for letting us know.