hamishcoleman/thinkpad-ec #151: Error: could not find any files in g2uj31us.iso.orig

leonurdo, 2019-12-15:
Hi,

I am having this (at the end of the text) error message.
yesterday I have updated my bois to the last good one BIOS 2.76 (having X230) but It seems Download x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)??

What is the problem? What should I do?

Thank you.

`leonurdo@debian:~/thinkpad-ec$` make patched.x230.img
Downloading x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW)
2019-12-16 00:21:12 URL:https://download.lenovo.com/pccbbs/mobiles/g2uj31us.iso [33927168/33927168] -> "g2uj31us.iso.orig" [1]
scripts/checksum --mv_on_fail g2uj31us.iso.orig g2uj31us.iso
./scripts/ISO_copyFL2 from_iso g2uj31us.iso.orig x230.G2HT35WW.s01D3000.FL2.orig 01D3000.FL2
Syntax error at line 5 for drive A: column 9 in file /etc/mtools.conf: unrecognized keyword
Error: could not find any files in g2uj31us.iso.orig matching 01D3000.FL2
make: *** [.d/generated.deps:81: x230.G2HT35WW.s01D3000.FL2.orig] Error 1
`leonurdo@debian:~/thinkpad-ec$`

leonurdo, 2019-12-16:
I think I figure out the reason of syntax error; my debian language is turkish and I think this cause the problem.
[((https://bugs.launchpad.net/ubuntu/+source/mtools/+bug/1479152))](url)

I will try to change my debian to english first by following these steps ( If I can not able to this I reinstall as default English) [https://superuser.com/questions/975138/change-debian-language-to-english](url)

But how about different BIOS version that tries to download? Mr. hamishcoleman replied before as this post to nil-0 who has experienced same problem of mine `https://github.com/hamishcoleman/thinkpad-ec/pull/130#issuecomment-516865247`

do I understand is it safe to continue with downloaded x230 BIOS 2.75 even I my X230 BIOS BIOS 2.76? because of they both use EC 1.14.

hamishcoleman, 2019-12-16:
Hi @leonurdo , that mtools.conf problem does indeed look like a language issue - especially annoying since the /etc/mtools.conf file is completely default and untouched by you or the thinkpad-ec software.

As mentioned in the earlier ticket, it is safe to continue because the EC versions are matching.  Also, the Lenovo BIOS release 2.76 appears to have been built by them with the wrong contents, so it is impossible for us to use.

leonurdo, 2019-12-16:
> Hi @leonurdo , that mtools.conf problem does indeed look like a language issue - especially annoying since the /etc/mtools.conf file is completely default and untouched by you or the thinkpad-ec [...]

Hi @hamishcoleman , 

Sorry my poor English , I am a little bit confused because you says it is safe and than it is impossible. 

please let me clear;  

I have updated my BIOS to the recent last good version BIOS 2.76 before as described in your instruction. as except syntax error make patched.x230.img gives me BIOS 2.75 instead of BIOS 2.76 as seen the code above.

so you says that continue with x230 BIOS 2.75 (G2ETB5WW) EC 1.14 (G2HT35WW) to create a bootable usb and boot my BIOS 2.76 system with this patched BIOS 2.75??? 

at the end of this proccess what will be my final BIOS version?

hamishcoleman, 2019-12-16:
First of all, yes, you can use the patched EC firmware built using the the BIOS 2.75 download.  This will work fine.  As you noted, the EC version is matching with your current installed BIOS version 2.76.

After patching your EC with this, your BIOS will not change - it will still be BIOS 2.76.

Patching your EC does not make any changes to the BIOS, just to the firmware in the Embedded controller - so you will not be downgrading the BIOS version.

Finally, my confusing note was intended to just be a small explanation - it is not possible for me to update this software to remove the confusion with the two different BIOS versions because there is a technical error that has been made by Lenovo - this does not affect you, nor does it affect using the patched EC built here.

leonurdo, 2019-12-16:
I have been shocked when I first saw your compability table update 1 hour ago. Because an x230 with the X220 keyboard was my dream computer and I finally found a clean X230 that made my dreams come true. It had older bios version and then I carefully followed the instructons you described. and I have updated bios the final last good one 2.76. I was sure that on compability table it was 2.76 but then list turned 2.75  I could not believe my eyes.. I hope that I do not turn my x230 a brick and can patch succesfully. Otherwise I will sell my new X230 i7 to able to hope to find another older bios.. 
I will try it when I return to home.
Thank you very much @hamishcoleman

leonurdo, 2019-12-17:
Yes it worked perfectly. both keyboard and battery patches. 
Thank you @hamishcoleman

closed 2019-12-17 by leonurdo