hamishcoleman/thinkpad-ec #152: is it possible I burnt the middle button?

peddy22, 2019-12-24:
I did not tape the correct side of the keyboard connector, now 100% of the keys work minus the trackpoint middle button. Is my keyboard fried? I've tried doubling up the padding on the inside to try and make a better connection and used thin stickytape. 

Thanks!

peddy22, 2020-02-19:
I would appreciate feedback on this matter

dkogan, 2020-02-29:
Do you figure anything out here? I now have 3 keyboards: 2 have everything working, except the 3 mouse buttons at the bottom of the keyboard, and the 3rd has working mouse buttons, but the entire middle row of keys doesn't work.

peddy22, 2020-02-29:
It is probably because it was taped incorrectly and burnt a line.

dkogan, 2020-03-01:
So I figured it out. I was using electrical tape for the pin isolation, but this was too thick, and was interfering with the connectivity of the adjacent pins. Switching to a thinner isolating tape fixed everything. The trackpoint buttons work now. Scotch tape works. It's not intended to be used for this purpose, but hopefully it won't degrade over time.

peddy22, 2020-03-02:
I just got a 45N2211 x220 keyboard and I doubled up on the foam rectangle underneath the retaining bracket, after a column of keys didn't worked. I taped it using a small piece of scotch tape.

KresoDenis, 2020-09-19:
Similar case here. Correct orientation and side. But there are 2 layers. I taped 3 pins on the first layer instead second. Burning smell was there, cant notice any burn hole. Later I removed the tape and put back together. Everything works but middle button is broken. Anyways I have replacement order from earlier since this language i currently have is not english. So hopefully once it arrives ill give it a second try. This time reading details :D

mmuman, 2022-04-20:
I also had the 3 trackpoint buttons not working on one of the modded keyboard until I reopened it and noticed the foam was too thin, so I added some rubber-like pad in-between it and the membrane and now it all works.