hamishcoleman/thinkpad-ec #154: EC 1.15 and !vyrain bios

propep, 2020-01-17:
Hi, now that there is 1virain bios, there is some chance that we can patch EC 1.15??

hamishcoleman, 2020-01-17:
I assume you are talking about https://github.com/n4ru/1vyrain/ ? that project looks interesting - thanks for mentioning it.

Unfortunately, the only way to flash the EC is to use the official flasher, which will still check the embedded signature - the 1virain software says it doesnt touch the EC at all, so I dont think that situation will change.

propep, 2020-01-18:
I asked because I was in trouble a little bit.

I had 1vyrain and EC 1.14, with only the battery patch (so yes, "This will NOT modify your EC"). however, today, after changing the keyboard, I also wanted to flash the patch for the keys, and at least for me the second part of the statement "You are safe to flash your EC ... at any time ..."did not work.

Then I mess up everything, with a wrong logical reasoning, and a little distraction, I found myself with 1vyrain and original EC 1.15 .. LOOL in 30 seconds neither the battery was working.

Long story short, I hoped that a downgrade (previously used to match the 1vyrain requirement) would also flash the ECs, and it worked (now not even the wifi XD was working with original bios) so I started again --> I) patches --> II) 1vyrain

I wrote all this, just in case another one, as noob as me found him self without battery, wifi and key XD
Anyway thanks so much for your work, and since I have mentioned  the "jailbreak" many times thanks also to n4ru.