hamishcoleman/thinkpad-ec #155: Got a x230 with a 2.77 BIOS, can I still patch it after I do an external flash with coreboot?

agyild, 2020-01-25:
Unfortunately I have prematurely upgraded my x230's BIOS to 2.77. I am going to external flash coreboot using Skulls very soon. I just want to be sure if it is possible that way to bypass 2.77 digital signature lock. Thanks.

hamishcoleman, 2020-01-26:
Once you install coreboot, there is no method to flash the EC firmware - If you want to have EC any patches installed, you will need to apply that beforehand.

The Lenovo BIOS settings screens should have a disable secure rollback prevention option that will allow you to flash an earlier version.

You could also just try applying the EC patch without downgrading - but nobody has reported doing that and I dont know what the results will be (Assuming they have written the software properly, it will just fail with an error message)

exander77, 2021-03-27:
@hamishcoleman So this means that I have to revert back to stock UEFI/BIOS, patch and install Coreboot?

Why can't be done using Coreboot? Does something need to be implemented in Coreboot?