hamishcoleman/thinkpad-ec #158: Goodbye to aftermarket x220t batteries on x230t laptop with BIOS 2.75 (GCETB5WW) with EC 1.15 (GCHT26WW)?

mokalovesoulmate, 2020-03-02:
As subject,

I stupidly updated everything after installing Windows 10 with Lenovo Vantage (Lenovo Vantage software also updated the BIOS even I know my battery were broken and planned to buy later)

Now I want to buy a x220t battery to be used on my x230t, then I arrived to this Github repository. I feel screwed up regarding the EC.

Is the genuine battery only the way to solve my x230t battery problem? (I havent make purchase to the x220t battery yet)

Thanks.

(currently sad, USD 220 wasted lol)

mszumilak, 2020-03-02:
With 2.75 bios for x230t only the original battery is going to work. The other option is downgrading back to 2.73 and installing patched EC.

mokalovesoulmate, 2020-03-02:
Wait, is it possible to downgrade BIOS from 2.75 to 2.73? That does include downgrading the EC?

mszumilak, 2020-03-02:
Yes, I did similar downgrade for x230 (2.77 -> 2.75) and yes, it will downgrade the EC as it is integral part of the UEFI package.

mokalovesoulmate, 2020-03-02:
Incredible, thanks. Will try

mokalovesoulmate, 2020-03-02:
Wohoo!
I downgraded the BIOS, created the custom BIOS, flashed custom BIOS and EC, now I am confident to buy a new battery!

closed 2020-03-02 by mokalovesoulmate