hamishcoleman/thinkpad-ec #159: Keeping Descriptions.txt up to date

samveen, 2020-03-06:
First, thank you for the framework you've written. It's so well designed such that I'm planning on using it as a sub-module to automate 1vyrain builds via travis-ci, instead of trying to build my own poor copy of it. With that in mind I have a couple of questions.

- Do you expect to track and update the BIOS and EC release versions and hashes into [Descriptions.txt](../blob/master/Descriptions.txt) whenever levovo releases new versions of BIOS and EC, or is it expected to be frozen, given CVE-2019-6171?

- I'm hoping to extract the BIOS file instead of the EC file from the original ISOs? Would you consider a PR that adds such a script (something similar to [ISO_copyFL2](../blob/master/scripts/ISO_copyFL2) like, an `ISO_extractFL1`)?

hamishcoleman, 2020-03-06:
Hi, thanks for the praise - and glad you think it is well designed.  I often have looked at it and wondered if anyone could follow my esoteric makefiles and helper scripts.

I was not planning on diligently tracking the new BIOS release versions - given that there was no ability to patch them, however it is a simple job to merge new sha code pull requests into the descriptions.txt, so I have no reason to freeze it completely.

I would certainly consider an extractor for the FL1 file.  The FL1 is a EFI capsule, with well defined extraction processes - so it would be actually simpler to make a generic one than the mess of options found in all the FL2 files

samveen, 2020-03-09:
It's so awesome to be able to run the following and get the correct ISO:

```
grep -v BROKEN Descriptions.txt |awk '/ '"$MODEL"' BIOS /{fname=$1; ver=$5} END{print fname,ver}' |xargs -r bash -c 'echo "Getting ISO of BIOS ver $1 for $MODEL"; make ${0}.orig'
```

I say well designed, because this works to get the latest BIOS ISO for every model supported by thinkpad-ec, just by setting the correct model.

I'll hopefully get the PR in within a couple of days.

hamishcoleman, 2020-04-08:
I just noticed that I had not answered the second half of your last message:

The format of the FL2 file changes occasionally, which means that the DOSFLASH.EXE utility also needs to chage to match, so both those files would need to be retrofitted into the older ISO to make it work.  However, that is not the biggest issue.

The BIOS makes undocumented calls to the EC, so if there is a mismatch between the BIOS and EC versions then you run the risk that the BIOS is going to make a call that the EC doesn't understand (or that the EC returns results that the BIOS does not understand).

Since this is a core critical part of a working laptop and could lead to bricking, experimentation in this area is quite risky.  There are mitigations that you can take to avoid being completely bricked (Like ensuring you can attach a JTAG programmer to your EC and flash a working version), but they are not simple.

Finally - and possibly most problematic - I assume you want the current BIOS version (for any bug fixes) and the old EC version (for the patches).  This combination is completely impossible, according to the release notes, since the mitigation for CVE-2019-6171 that caused the recent BIOS release is to stop you from flashing patched EC versions.  The good-ish news is that there are no other known updates to the BIOS with that latest version, so you are not missing out on any bug fixes.