hamishcoleman/thinkpad-ec #160: drives not showing up using "lsblk"

BrianKiw007, 2020-03-06:
hey all i'm trying run this patch for t430s using  Ubuntu 18.04 LTS... when i try to see what drives are their using "lsblk" :~/thinkpad-ec# lsblk
lsblk: failed to access sysfs directory: /sys/dev/block: No such file or directory

hamishcoleman, 2020-03-07:
Thats a really strange issue to have - it sounds like there is something not right with your Linux setup.  What distribution are you using?

As a cruder alternative to lsblk, you can try "cat /proc/partitions"

BrianKiw007, 2020-03-07:
i'm new the Linux i just used the one from Microsoft  version Ubuntu 18.04 LTS... i tried that comand also got back the following /thinkpad-ec# cat /proc/partitions
cat: /proc/partitions: No such file or directory

m-kozlowski, 2020-03-07:
> Microsoft version Ubuntu 18.04 LTS
wait, what?

so:
-you're new to linux
-you're not using linux
-you've encountered bug that is most likely a lack of Microsoft's WSL compatibility

and you reporting it to embeded controller firmware patcher issue tracker?

hamishcoleman, 2020-03-07:
To be fair, they are advertising it as "just like real linux" and do say "this is Ubuntu", so if you are new to Linux, the marketing might make it all sound reasonable.

That being said, you could should be able to build the patched image with in the WSL environment, it is just writing it out to the USB stick that needs hardware specific details.

Perhaps you could use something like etcher.io from the windows side to write the image file to the USB stick (but I have no clear directions, have never done this, and probably couldnt help you)

Kokkolis, 2023-05-12:
Hmm, it seems I am having a very similar problem. I know next to nothing about Linux (forced to download as the EC-Controller patch is only available through the Linux path).

So, I have downloaded Pop!OS (an Ubuntu derivative based on Debian as I understand). Have followed all commands, and then I get stuck at the "lsblk" command. 

No matter what USB stick I am attaching and no matter to which port, the "lsblk -d -o NAME,SIZE,LABEL" command, it always returns my two hard drives ("sda" with Windows 11, "sdb", which is split between Windows 11 storage space / Linux PoP!OS), and one with name "zram0" and size 15,3G.

I have tried many different USB's, of different capacities, from 8GB to 32GB, I can only see "zram0". One of the USB's is formatted as NTFS for windows, another one is bootable and contains the Live distro of PoP!OS, bought another one just to try, formatted as FAT (and not re-formatted since its purchase).

If I try to substitute the "sdx" with "zram0", the instruction returned an error (record 1, record out 0, 0/0 bytes copied)... hope I haven't damaged my hard drives.

All USB's show perfectly in Windows 11. 

What am I missing? When I stick any of these USB's, I cannot see any "pop-up" messages in Linux the way I see in Windows. Running command "lsusb" in Pop!OS terminal, before and AFTER inserting the brand USB, shows my USB in its list of devices found, it shows as "Device 42, ID xxxxxxx, SanDisk Corp. Ultra", please see attachments.

Any help is much appreciated as I am now stuck at the ultimate command instruction...

Thank you
Christos

[image]
[image]
[image]