hamishcoleman/thinkpad-ec #163: Random hibernation with reason "Battery"

peddy22, 2020-03-15:
Event viewer shows:

Critical Battery Trigger Met
followed by

The system is entering sleep.

Sleep Reason: Battery

This is on 100% battery (X220 battery), plugged in, on the X230 with the ec modded. 

I'm using the latest Windows 10.

Can anyone lend a hand?

peddy22, 2020-03-21:
I also get "This battery is very low" warnings on 100% battery

peddy22, 2020-04-03:
@hamishcoleman can you chime in on this?

hamishcoleman, 2020-04-04:
I dont know that I have any light to shine here.  It could be that your battery has died - do you have access to an x220 to test it in?

Is it actually showing 100% charge?  

If you dont know about the windows battery report command: "powercfg /batteryreport" it might be worth running that and looking at the html output file it generates to see if there is any further information.

peddy22, 2020-04-04:
Hamish, thanks for the reply.

Yes, it often does this while plugged in/100% charge (at least before I
disabled the shutdown on critical battery action), but there is no such
option for on-battery mode. By often I mean maybe one or twice a day.

The battery is pretty healthy at about 80% although I suspect it is a
compatible rather than original.

There is sometimes a message in event viewer about the Embedded Controller,
I'll try and catch it and post it here next time that it happens.

I will have an X220 in a couple of days I could test it in.

Kind regards

Arno

nmusatov, 2020-04-05:
Have absolutely the same issue on vanila Thinkpad P51 without any mods.
So probably it is not an issue of this project.

peddy22, 2020-04-07:
Fascinating. I wonder if I'm due for a Windows reinstall. Cheerio.