hamishcoleman/thinkpad-ec #165: T530 bat mod

Jaggie88, 2020-03-17:
hello there,

is it possible to use the w530(or perhaps t430) files from img.d for T530 model and create ISO with those files?
I only need the battery MOD... keyboard will stay unchanged

hamishcoleman, 2020-03-17:
I'm not sure what the problem you are trying to solve is - there is already support for the t530 in the build system, so you should be able to just use it following the normal instructions

Jaggie88, 2020-03-17:
ok thank you very much for such quick reply, i was just now looking in the img.d folder and got confused that there are no files...
so I should just build the ISO, boot in into flash disk and it should give me option for t530?

hamishcoleman, 2020-03-17:
I'm not sure where you are looking, as I see files in the t530.G4HT39WW.img.d/ directory.

You need to build the correct ISO for your laptop - once booted, there are no options.

You should follow the steps at: https://github.com/hamishcoleman/thinkpad-ec#step-by-step-instructions
and be sure to configure which patches you want (before step 6) using the instructions at: https://github.com/hamishcoleman/thinkpad-ec/blob/master/docs/CONFIG.md#configuring-which-patches-are-used

Cheers!

Jaggie88, 2020-03-17:
understood :)

Jaggie88, 2020-03-17:
done... 
at first I had to donwgrade the BIOS version with the help of .bat from 1vyrain project(because a while ago i applied his patched BIOS to 'cure' the wifi whitelist), then I followed the fairly simple instrunctions(step by step section of your project) in Ubuntu running on VirtualBox, created bootable flash drive, and used only the battery mod to flash the EC, and again flash the 1vyrain BIOS(in the meantime I had to take out the wifi card, as it would not boot with it at all)

I have to thank you very much...
the process is very well documented

Chiqui1234, 2020-03-21:
Hi guys, I have the battery problem. In my case, I have a genuine 44++ working since 2019, but 2 days ago my X230 says "battery is not supported" bla bla bla...
I made the booteable pendrive thanks to thinkpad-ec, flashed the notebook but the battery alert still there, so I can't charge the battery.
Options?
Thanks in advance

Chiqui1234, 2020-03-22:
> Hi guys, I have the battery problem. In my case, I have a genuine 44++ working since 2019, but 2 days ago my X230 says "battery is not supported" bla bla bla...
> I made the booteable pendrive thanks [...]

Sorry, now I read this and my battery is actually charging! https://github.com/hamishcoleman/thinkpad-ec/blob/master/docs/CONFIG.md#configuring-which-patches-are-used

I have the original X230 keyboard, so:
$ make patch_enable_battery clean
$ make patch_disable_keyboard clean
$ make patched.x230.img

And directly to my pendrive stick. You have a Paypal link donation or BTC wallet? :D
You save me with another oldie Thinkpad from my brother haha

hamishcoleman, 2020-04-04:
Thanks @Chiqui1234 - it is always good to hear of successes!

I dont have a donation link, but I'm sure my local hackspace would love to have a donation: https://www.dimsumlabs.com/ has a spot donations box on the side

guille1978, 2021-02-16:
I am using the linux mint 17 version, when executing the command: sudo dd if = patched.T530.img of = / dev / sdc1, the usb drive is empty and unformatted, what would be the problem?

hamishcoleman, 2021-02-16:
It looks like you are using a partition device, not the whole disk - if your usb stick is /dev/sdc then you should leave the "1" out of the command.