hamishcoleman/thinkpad-ec #166: L440 Battery Whitelist

ryangweb, 2020-03-27:
Hello Hamish

What sort of changes would I need to make this work for a L440?

(During the Covid-19 isolation I have decided to dig out my underused Lenovo L440 Thinkpad. My plan was to wipe Windows 7 and install Linux before gifting it to my daughter.  The battery was in a good state but is no longer recoqnised as genuine.)

hamishcoleman, 2020-04-04:
Hi @ryangweb , sorry for the slow reply - the same Covid stuff has thrown me quite a lot of chaos!

I had a quick look at the L440 EC stuff, and it is not using the same architecture as the xx30 series.  I should be able to add some more support for extracting the EC image from the downloaded firmware and you can have a poke around in that to see what is possible.

- It looks like it is using a EC with a 8051 CPU, so there are lots of disassembly tools to help with finding the battery check code.
- The EC firmware is included in a UEFI capsule update
- I have not looked at how the EC flashing is done, but it is probably done as part of installing the entire UEFI capsule file
- This probably means working out some checksums and signatures for signing the patched capsule.

hamishcoleman, 2020-04-08:
I have update the repository with some changes to allow extracting the EC firmware for the l440 (See 084a8f94f4a453d0e1c9643bef291b3d9209871b)

There is no support for patching the firmware, but at least you can have a look at it and see what might be involved.

you should be able to `make l440.J4HT30WW.img.orig` and get the EC firmware

ryangweb, 2020-04-08:
Thank you very much Hamish. Sorry I didn't reply to your earlier message.

anselmusr, 2024-12-26:
how to fix "the battery installed is not supported by this system and will not charge" thinkpad l440?