hamishcoleman/thinkpad-ec #168: Fix/adjust polling rate for Thinkpad keyboards?

RansomIblis, 2020-04-08:
As per https://www.reddit.com/r/thinkpad/comments/9ivh0u/thinkpad_keyboards_measurement_of_polling_lag_and/ -- is it possible to patch the keyboard firmware to eliminate the polling issue that has been plaguing Thinkpads for the better part of a decade? Or am I totally misreading what this program can do?

hamishcoleman, 2020-05-29:
I've not experienced this lag, and I have been using the thinkpad keyboards for significantly more than a decade.

That being said, it might be a hardware limitation - and thus not possible to fix (if it was due to the shape of the keyboard scanning, or the speed of the matrix scan built into the MEC controller chip).

This repository certainly has all the tools you would need to start looking for answers, so you have not misread that - but it is not currently containing the simple answer you are looking for.

dm17, 2021-02-15:
Should be easy... Lenovo has been very lazy on basic fixes their "enthusiast community" is constantly writing about everywhere. If you want "best laptop" keyboard, then you want a higher polling rate; simple!