hamishcoleman/thinkpad-ec #169: Fatal error building x230 patch @ [Makefile:296: mec-tools/Makefile]

ginto37, 2020-04-24:
Tried to build the patch on two separate platforms recently, but got this same error. I had the latest available clone of the repo at the time. Live Internet connection on both machines during the build:
`make patched.x230.img`
`Downloading x230 BIOS 2.75 (G2ETB5WW)   EC 1.14 (G2HT35WW)`
This part succeeds, I have the ISO files now.
`git submodule update --init --remote`
`fatal: not a git repository (or any of the parent directories): .git`
`make: *** [Makefile:296: mec-tools/Makefile] Error 128`
On the other platform the error was `129` instead of `128`, but the rest was the same.

hamishcoleman, 2020-04-24:
This does sound like you have a problem with the internet, or the version of git you have installed.

Can you try running some tests and paste the output:

```
git --version
git status
git submodule status
git submodule update --init --remote
```

ADHSoft, 2020-08-17:
That happens when you download the code as a .zip on github.com , instead of running the git clone commands.