hamishcoleman/thinkpad-ec #170: X230i jail break

ageryun, 2020-04-30:
I had below screen shot while I was in jail break of my thinkpad x230i.

[image]

bios was downgraded to 2.60 from 2.77. but still has same 'unsupported' message.

hamishcoleman, 2020-04-30:
Hi ageryun,
Are you reporting this issue to the right project?  Nothing on your screenshot looks like you are trying to patch your EC firmware.

ageryun, 2020-04-30:
Hi, I was trying to jail break bios by using Ivyrain ISO image flashing. but not sure if that was done correctly. If here is not for this kind of question, then sorry I will delete that posting.

hamishcoleman, 2020-04-30:
It is not so much that you shouldn't post it here - more that you will reach the right audience if you post an issue in the Ivyrain github

ageryun, 2020-04-30:
Oops. Sorry. Thank you for letting me know. My first time to post on Github. Can you delete my issue..? I tried but couldn't.

closed 2020-04-30 by ageryun

hamishcoleman, 2020-04-30:
No need to apologise.  Also, no need to delete the issue - anyone else who finds this will be able to see and follow the same advise :-)