hamishcoleman/thinkpad-ec #171: Add wget to the command to install the prerequisites

zacchi4k, 2020-05-14:
Running this on WSL2, I was getting an error because wget was not installed, so I suggest to add wget to the command to install the prerequisites

hamishcoleman, 2020-05-14:
Thanks for pointing that out.  I have added wget.

Also, you will probably find other issues using WSL if you try to write the image to a USB stick

zacchi4k, 2020-05-14:
> Thanks for pointing that out. I have added wget.
> 
> Also, you will probably find other issues using WSL if you try to write the image to a USB stick

For writing the images I just copy them over to the Windows folders (/mnt/driveletter/etc) and write them with Rufus or balenaEtcher

hamishcoleman, 2020-05-15:
Good to know that it is that simple.  Thanks for testing!

closed 2020-05-15 by hamishcoleman