hamishcoleman/thinkpad-ec #172: libssl-dev dependency not available on arch based distros

seanluse41, 2020-05-19:
Hello. Fantastic tool here, that I'd love to use, however arch and manjaro are unable to satisfy the libssl-dev dependency, as far as my research tells me.
It seems that on arch etc things are handled with openssl.
Is there anyway to have support for openssl?
Thanks again

hamishcoleman, 2020-05-19:
The Debian "libssl-dev" package is just the development libraries separated from openssl.  It looks like the arch openssl package contains all the dev parts by default, so you should be good to use that.

seanluse41, 2020-05-19:
Amazing response time!
Apologies, I thought an error I was getting was due to not having the dependencies satisfied.

```
[sean@x230 thinkpad-ec]$ make patched.x230.img
git submodule update
make -C mec-tools
make[1]: Entering directory '/home/sean/thinkpad-ec/mec-tools'
cc -Wall -O2 mec_csum_boot.c -o mec_csum_boot 
make[1]: cc: No such file or directory
make[1]: *** [Makefile:8: mec_csum_boot] Error 127
make[1]: Leaving directory '/home/sean/thinkpad-ec/mec-tools'
make: *** [Makefile:300: mec-tools/mec_encrypt] Error 2
```

hamishcoleman, 2020-05-19:
Ah! - yes, that error message is not a ssl library issue.

You are missing a C compiler - the Debian package "build-essential" installs gcc automatically, but on arch, you probably need to install "gcc"

seanluse41, 2020-05-19:
Seems to be working now. I've changed the title, so others in the future can search for this. Thanks again.

closed 2020-05-19 by seanluse41

hamishcoleman, 2020-05-19:
Good news!

Thanks for persisting :-)
(Its not hard to get good response times these days - Its not like I'm going out...)