hamishcoleman/thinkpad-ec #179: Flashing back the stock firmware?

dustojnikhummer, 2020-08-02:
I have had a 7-row on my X230 for about a year and a half but I need to put the original keyboard back as the 7-row is broken (drifting trackpoint, trackpoint scroll does not work etc).

I'm running BIOS 2.72, dated 2018-04-11. I don't have the original files as I did the flash in Live Ubuntu.

What do I have to do in order to get the stock keyboard functionality back? And can the battery unlock be kept?

hamishcoleman, 2020-08-03:
You can simply build a new image with the keyboard patch disabled but the battery patch enabled. You can check https://github.com/hamishcoleman/thinkpad-ec/blob/master/docs/CONFIG.md for some more details.

The EC version included with your BIOS is the same as that used by the current version of this repo, so that is still compatible.

dustojnikhummer, 2020-08-03:
I found that, but the documentation is so luck luster it is not even funny. Sure, there is “Use this command”, but there is no when, where and in what syntax to use them. In the end running those commands (battery on, keyboard off) BEFORE running the ISO build worked. 

In the end it worked. But I’m surprised a BIOS update did not reset the EC to the default Lenovo setting.

hamishcoleman, 2020-08-03:
I think you could have phrased that more politely.

However, I have made an update to the readme, perhaps you can tell me if this has increased the 'luster'

closed 2020-08-03 by hamishcoleman

dustojnikhummer, 2020-08-03:
Sorry if I came out as an a-hole, but I actually got angry at it. I'm very thankful that you are helping the community with this for free, but I tend to get frustrated when something quite important gets brushed off "Yeah, just use this" but there is no explanation how to actually use _that_  yet other categories get more explanation than they should.

Also one thing, I think the "Ensure your BIOS has been configured to boot from "Legacy" and not "UEFI" before trying to boot."should be right above the "Booting the USB stick" and maybe in Bold.

Thank you!