hamishcoleman/thinkpad-ec #180: Keyboard language

toniz4, 2020-08-06:
Does this patch for the x230 works for all languages? or just english?

hamishcoleman, 2020-08-07:
Because we are simply adding some missing bits to the existing keyboard table, any international keyboards should work fine.

However, I have not personally tested that..

Which keyboard layout are you hoping to use?

toniz4, 2020-08-07:
I hope to use a abnt2 Brazilian keyboard, but i'm worried there will be compatibility problems. And the original keyboard on the x230 is a abnt2 keyboard too. I think if a iso keyboard works, possibly a abnt2 keyboard would work too, since abnt2 is just a iso layout but with 'ç' and a '~' and accent button re positioned.

hamishcoleman, 2020-08-08:
Ah, yes - this is just a ISO layout with a couple of different keycaps so you should be fine

closed 2020-08-08 by toniz4