hamishcoleman/thinkpad-ec #181: T440p battery hack

jlelusz, 2020-08-07:
I spent last week trying to find a way to get rid of unauthorized battery not charging in T440p. I managed to find fixes for many Lenovo laptops, both pre-T440p and post-T440p, but nothing for my specific model.
Weird thing is that I've had the non-Lenovo 9-cell battery for a year and it worked flawlessy until a recent Window and Lenovo Vantage update around 2 weeks ago. Result - laptop refuses to charge the battery.
They must've changed something in EC firmware, the warning pops up after Bios and before Windows starts. I downgraded Bios to something from 2 years ago (downloaded old Bios from Lenovo) and it did change the EC version - but it still refuses to charge the battery, which is strange. Tried removing CMOS battery for 1/2 day - no joy.

What would need to happen to make a fix/patch for T440p? I'm happy to help - we have more JTags in the office than coffee cups, so happy to read/write EC on my T440p for the greater good ;-)

jcholsap, 2020-08-08:
It does sound like the EC firmware. I've flashed a few T540p mbds. While investigating, didn't find a JTAG connector near the MEC. There was a nearby edge connector but I didn't trace it out. I remember thinking I'd need to get out my soldering iron and find some good contact points. BTW, had to remove the T540p mbd. Some folk are uncomfortable with removing screws. Anyway, documenting a JTAG connection could be helpful. Also, pressing F1 at boot and looking at the third line of text gives the EC version. Or maybe someone has an idea for pushing an EC downgrade.

hamishcoleman, 2020-08-08:
The T440p firmware that we can download and extract is encrypted - so getting a clean dump from the physical hardware might allow us to find the encryption keys (in the same way zmatt did for the x230).  From a brief look, the firmware appears to be similar in layout to the older xx30 firmwares, so we might simply be able to port the patch forward.

You can see the current firmware extraction process with: `make t440p.GLHT30WW.img.orig`

A quick online search finds a schematic for the T440p, which shows that there should be an unpopulated JTAG header for the EC, and identifies some resistors that might need to be populated to enable the jtag port.

jlelusz, 2020-08-08:
OK gents, that's really great feedback! I'm off for holidays for two weeks (family duties). When back, I'll try to read&post the EC firmware. Will try to find the correct pins of the the correct JTAG to read the FW. I'll do my holiday research, looking at PCB pictures etc, assuming it's MC 1663.

Given that the battery worked for a year and only then threw an error, I thought maybe a dud battery rather that change in BIOS/FW is the case - but you never know.

jlelusz, 2020-08-27:
Not long now, everything is soldered (see photos). Hopefully in the next few days I'll be able to read the flash. 
[image]
[image]

jlelusz, 2020-09-01:
GLHT25WW attached.

[t440p-1.zip](https://github.com/hamishcoleman/thinkpad-ec/files/5159151/t440p-1.zip)

I've read the Flash a few times, all files were the same so I'm assuming that eventual bit errors (due to dangling wires) are unlikely.

renoxtv, 2023-03-24:
Hi, sorry for posting on a 2 year old thread. If its possible  and if you still have your T440p, could you please send a more detailed picture of the MEC1633L area?