hamishcoleman/thinkpad-ec #183: battery patch did not work with my X230

AmjadKh, 2020-10-29:
I have lenovo thinkpad x230. I updated the bios to the latest version using lenovo update tool for windows (2.77 for bios and 1.15 for EC). And then, the battery is not able to be charged. It shows in windows: "connected, not charging".

So I started to search, and found that lenovo prevents using non genuine batteries in the last update. 
And I found this github project that solves the problem by applying a patch.

I read the docs, and I did a successfully downgrade the version of bios to 2.75, and the EC to 1.14.  see the image below:
[image]

and then I tried to apply the patch using the instructions. I also used this command to apply the battery patch only:

`make patch_disable_keyboard clean && make patch_enable_battery clean`

(I get this command from this page: [https://bystram.be/posts/thinkpad-x230-battery-ec-patch/](https://bystram.be/posts/thinkpad-x230-battery-ec-patch/))

and when I tried to apply the patch, I see the instructions screen and I continue and everything runs normal, no errors. 
[image]

Then the laptop reboot by itself and shows: "Flashing Embedded Controller... Please do not power off" and then reboot again by itself.

By then, I assumed that the problem is solved. But again, the battery cannot be charged. I tried to shutdown and removed the battery and plugged it again, but didn't work.

I noticed also, when the laptop is powered off, when I plug the charger, the battery led icon on the laptop cover, it flashes for 3 times in yellow, and then turn off.

Is there anything I can do?

hamishcoleman, 2020-10-29:
Normally, the first thing I would suspect is that the BIOS had not programmed the new image to the EC, but you specifically note the "Flashing Embedded Controller... Please do not power off" message, which is showing that the image is being programmed.

The yellow flashes sound like the firmware doesn't like the battery - have you read the Lenovo documentation on that blink sequence? They are generally quite good at documenting what all the lights mean in the hardware maintenance manual.

Finally, it might be the battery pack itself that is dying.  The restrictions on the authentic battery are not new ones - in fact, I am of the belief that all BIOS/EC versions on the x230 have these restrictions - so if nothing else was changed, just upgrading the BIOS version should not have started rejecting the battery.  Perhaps it is worth looking at the battery details:

Generate some battery details for you to examine:
- Windows: ```powercfg /batteryreport```
- Linux: ```upower -d```

AmjadKh, 2020-10-29:
Thanks for your quick response.
Before the update, the battery was charging normal.
for the blinking 3 times, I read the docs and it says: 

> "The battery status indicator blinks three times: The ac power adapter is attached to the computer."
So, it says nothing about the battery, it means the AC power is attached.

I tried to use the commands you suggested. I have windows 7, so the `powercfg /batteryreport` didn't work, but there is another command `powercfg -energy` which output HTML file. Here is the important part of it:
[image]

Then I used a live ubuntu usb to run `upower -d` and its output is:

```
Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              Thu Oct 29 21:51:11 2020 (4112 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  power supply:         yes
  updated:              Thu Oct 29 22:59:11 2020 (32 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               charging
    warning-level:       none
    energy:              0 Wh
    energy-empty:        0 Wh
    energy-full:         0 Wh
    energy-full-design:  0 Wh
    energy-rate:         0 W
    percentage:          0%
    capacity:            100%
    icon-name:          'battery-caution-charging-symbolic'

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Thu Oct 29 21:50:29 2020 (4154 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               charging
    warning-level:       none
    energy:              0 Wh
    energy-full:         0 Wh
    energy-rate:         0 W
    percentage:          0%
    icon-name:          'battery-caution-charging-symbolic'

Daemon:
  daemon-version:  0.99.11
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  yes
  critical-action: PowerOff

```

I forget to mention that I did the downgrade in two separate steps. That's because the official method requires a charged battery and I don't have another one. 
So, first I downgraded the bios using Win98 DOS and using this command:
`dosflash.exe /sd /file G2ETB5WW\$01D3000.FL1`
and the second step downgrading the EC using this command:
`dosflash.exe /sd /ipf ec /file G2ETB5WW\$01D3000.FL2`

hamishcoleman, 2020-11-04:
Neither of those battery reports show any sign of detecting the battery.  If I remember correctly, a system that has detected a non-genuine battery will still show the serial number (and other details) in those reports.  Additionally, your boot up is paused with a message about the battery - which you have not mentioned seeing.

Based on those observations, unfortunately, I think that something is actually broken in your battery.

AmjadKh, 2020-11-05:
@hamishcoleman , There is no message on boot about the battery.
I will try to find another battery and test again.

vitlib, 2020-11-16:
@AmjadKh I'm doing the same thing (manually downgrading) and followed the same procedure as you, but my BIOS keeps showing 1.15. 
Could it be that your EC is still the old 1.15 and it just displays the wrong version number?
If you solve this, let us know!

vitlib, 2020-11-16:
Ok I managed to solve it by repeatedly switching from UEFI first to legacy first. Eventually it did the trick.

AmjadKh, 2020-11-17:
@vitlib it is good news that you solved the problem, I'm happy for you. 
Could you please mention the steps, what did you do exactly?

vitlib, 2020-11-17:
I switched the BIOS startup option from legacy to UEFI after flashing, then back to legacy again., then UEFI. 
It's reported in the readme that sometimes the process is not completed until you do this, and it worked

AmjadKh, 2020-11-17:
@vitlib do you mean after flashing directly?
I mean do I need to flash again? or just change legacy to UEFI as you mentioned?

vitlib, 2020-11-17:
Yes, after flashing (after the "Flashing EC") on boot. 
I flashed multiple times, though, so can't point you to the "correct" procedure.

kmeaw, 2020-11-28:
@AmjadKh, try changing the PSU. If the PSU does not supply enough power, EC and/or BIOS will not be flashed, and you will not get "Flashing EC" message on the boot. I had this issue on an i7 model, which requires a more powerful PSU, than i5 one. It seems that there is some circuitry inside the PSU which tells the EC the maximum available power, and the system refuses to run dangerous operations if there is a risk of shutting down due to over-current situation.

fuzzytrick, 2021-03-17:
Hey, 
@AmjadKh, have you managed to solve anything? Your issue is almost identical to mine, and I've tried everything, to no avail. 

I'm using an x230 as well, but with an i3-3120m processor. Battery is not recognized in Windows and nor in Linux. No message on boot about the battery compatibility. Also, the battery discharges. I reckon the flash did not go through. I used to have my bios version to 2.76, but I downgraded to 2.60 using IVPrep, and eventually updated to 2.75, trying to flash the EC Patch on both 2.60 and 2.75.

As kmeaw suggested, I even tried to change the PSU, having another x230 charger around, but the result was the same.
Thank you!

AmjadKh, 2021-03-17:
@fuzzytrick 
Unfortunately, the issue is still exist. I was planning to downgrade the version to lower than 2.75. But I was frustrating and didn't try.

rasteryaev, 2021-03-23:
I'm having a similar issue on the t430. I do everything correctly, and even apply the patch. I get the flashing ec message on reboot, but when I put my battery in, it's undetected. I'm using a duracell t430 battery which doesn't seem faulty in itself - it discharged just fine so it can at least hold a charge.

deanokn, 2021-04-12:
@rasteryaev 
Which BIOS version is on your T430? I stupidly deleted my friend's firmware he downloaded on this machine through a BIOS update and he doesn't remember so I'm looking for a fix that will work for a T430. This battery is just fine, given that 's been in use for 2 years with this laptop now.

SiamakMirifar, 2021-08-22:
Hi @AmjadKh,

I had the same problem as you, but I could solve it. I share it with you how could I do it (I took all these steps in Ubuntu):

In terminal:

1- `git clone https://github.com/hamishcoleman/thinkpad-ec`
2- `cd thinkpad-ec`
3- `make patch_disable_keyboard clean && make patch_enable_battery clean`
4- `make patched.x230.img`
5- `lsblk -d -o NAME,SIZE,LABEL`
!!! For this step be sure you are addressing correct USB to make Bootable patch for me it wa "sdb" it might be different for you
6- `sudo dd if=patched.x230.img of=/dev/sdb bs=4M status=progress conv=fsync` 

Now the USB is ready for booting. Go to the BIOS setting and change the boot to LEGACY. Reset and save. Now boot the USB and let it do it works. I think the POINT here was I just patched the battery not keyboard and ... . The first boot after patching it shows writing on the embedded (If I remember correctly). And after the system reboot it self again you can go to your BIOS and change the boot as you wish.

My BIOS version is: 2.73 (I downgrade it from the latest version. I did it in windows with the .exe file that was available in LENOVO website)

BacaR00T, 2023-03-07:
@AmjadKh Hi, I'm in the same situation as you were.. did you figured it out?

AmjadKhateebHu, 2023-03-07:
@BacaR00T, I replaced the battery. If all the above solutions did not work for you, then it may be your battery is dead and need to be replaced.

AmjadKh, 2023-03-07:
@BacaR00T , I replaced the battery. If all the above solutions did not work for you, then it may be your battery is dead and need to be replaced.

BacaR00T, 2023-03-07:
The system is recognizing it, after running "upower  -d" this is the outcome:

`Device: /org/freedesktop/UPower/devices/line_power_AC
  native-path:          AC
  power supply:         yes
  updated:              Tue 07 Mar 2023 12:43:07 PM CET (103 seconds ago)
  has history:          no
  has statistics:       no
  line-power
    warning-level:       none
    online:              yes
    icon-name:          'ac-adapter-symbolic'

Device: /org/freedesktop/UPower/devices/battery_BAT0
  native-path:          BAT0
  vendor:               LGC
  model:                45N1025
  serial:               6700
  power supply:         yes
  updated:              Tue 07 Mar 2023 12:44:37 PM CET (13 seconds ago)
  has history:          yes
  has statistics:       yes
  battery
    present:             yes
    rechargeable:        yes
    state:               pending-charge
    warning-level:       none
    energy:              58.86 Wh
    energy-empty:        0 Wh
    energy-full:         94 Wh
    energy-full-design:  93.24 Wh
    energy-rate:         0 W
    voltage:             11.676 V
    charge-cycles:       N/A
    percentage:          62%
    capacity:            100%
    technology:          lithium-ion
    icon-name:          'battery-full-charging-symbolic'
  History (charge):
    1678189387	62.000	pending-charge
    1678189387	0.000	unknown
  History (rate):
    1678189387	0.000	unknown

Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         yes
  updated:              Tue 07 Mar 2023 12:43:07 PM CET (103 seconds ago)
  has history:          no
  has statistics:       no
  battery
    present:             yes
    state:               pending-charge
    warning-level:       none
    energy:              58.86 Wh
    energy-full:         94 Wh
    energy-rate:         0 W
    charge-cycles:       N/A
    percentage:          62%
    icon-name:          'battery-full-charging-symbolic'

Daemon:
  daemon-version:  1.90.0
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  yes
  critical-action: PowerOff
`

But there is no way of charging it

SiamMiri, 2023-03-07:
It did work for me just as remember I tried for sometimes and it start working. Back then I had Ubuntu, and now I am on windows and still it is working

BacaR00T, 2023-03-07:
@SiamMiri Did you use just this patch or did you go with 1vyrain as well?

SiamMiri, 2023-03-07:
To be honest, I do not remember exactly as I did it more than one year ago, and I did not need to repeat the process. But I am quite sure I used methods and files from here and it works.

limaiem, 2023-04-22:
> Hi @AmjadKh,
>
> I had the same problem as you, but I could solve it. I share it with you how could I do it (I took all these steps in Ubuntu):
>
> In terminal:
>
> 1- `git clone [...]

thanks man, this helped me to make a 29++ batt work on an x230.

XeonNAS, 2025-01-20:
> Hi [@AmjadKh](https://github.com/AmjadKh),
>
> I had the same problem as you, but I could solve it. I share it with you how could I do it (I took all these steps in Ubuntu):
>
> In terminal:
>
> 1- `git [...]

Argh, having the same problem and when I try these commands it fails on the make, note I am only trying to make the battery patch:
`make -C mec-tools
make[1]: Entering directory '/home/john/thinkpad-ec/mec-tools'
cc -Wall -O2 mec_csum_boot.c -o mec_csum_boot 
make[1]: cc: No such file or directory
make[1]: *** [Makefile:8: mec_csum_boot] Error 127
make[1]: Leaving directory '/home/john/thinkpad-ec/mec-tools'
make: *** [Makefile:345: mec-tools/mec_encrypt] Error 2`

kmeaw, 2025-01-20:
@XeonNAS, looks like you don't have the development tools installed. If you are running Ubuntu or Debian, please try installing the "build-essential" package.

XeonNAS, 2025-01-20:
> [@XeonNAS](https://github.com/XeonNAS), looks like you don't have the development tools installed. If you are running Ubuntu or Debian, please try installing the "build-essential" package.

Thanks @kmeaw , running latest Ubuntu stable. Will install that package and retry make.

XeonNAS, 2025-01-28:
> [@XeonNAS](https://github.com/XeonNAS), looks like you don't have the development tools installed. If you are running Ubuntu or Debian, please try installing the "build-essential" package.

Ok, added that package and went through the process, applied patch and still 0% and not charging with new battery........will just live with it.