hamishcoleman/thinkpad-ec #186: Change fan speeds

pdpapper, 2020-12-07:
Would it be possible to modify the default fan speeds with an EC mod?

I know you can use thinkfan or tpfancontrol to change how the fan levels respond to temperature, but my question is about the actual speeds associated with each fan level.

On my T430s, fan level 1 is about 3000rpm and loud enough that you can hear it if you listen for it. With the exact same fan/heatsink in my T420s, level 1 is about 2000rpm and can't even be heard in a quiet room. I also noticed that there is no difference between level 6 and level 7 on the T430s. So it seems like level 1 on the T430s could be set to ~2000rpm like the T420s and then all the other speeds could just be set 1 higher, which would still give the same existing RPM options, but take advantage of all the levels from 0 to 7.

Is this possible and something that the EC controls?

digmorepaka, 2021-01-20:
Not directly related to your question, but multiple different manufacturers of fans were used with different speeds and blades even within the T430s. It is highly likely that the fan is different between your T430s and T420s.

pdpapper, 2021-01-23:
It's the same physical fan. I just swapped the T430s motherboard into my T420s case.

I ended up creating a fan profile that keeps the fan completely off below 70*C and I use that when I want quiet. It almost never comes on at all that way, so I guess it's pretty good.

I'd still love it if it was possible to have the extra quiet fan speed setting, though.