hamishcoleman/thinkpad-ec #196: x230 keyboard issues

gulafaran, 2021-03-23:
Hi ive gone through the various FN + key combos and found some irregularities however not sure if they are fixable or meant to be like that but i thought il make an issue for it atleast.

FN + F2  generates XF86Battery 
FN + F3 generates XF86ScreenSaver

which is reversed to how the symbols on my 7 row keyboard is. checked with xev on linux.

FN + space, FN + F12 , neither generates an keypress/event. 

the rest is seemingly working as intended besides the caps lock led but thats what i understand unfixable.

florhizome, 2021-07-05:
Hi, 
I‘m not at all an expert on this in general but for the problem of keys existing but not in the right position / being reversed you can change their .hwdb settings, if that was the problem.

pavelselivanov, 2022-02-11:
For me disabling keyboard patch worked:
make clean
	
make patch_disable_keyboard
make patch_enable_battery 

make patched.x230.img

This way I got image that allows to install non Lenovo battery that charges and keyboard works fine.