hamishcoleman/thinkpad-ec #203: L430 with BIOS G3ET68WW (2.07) / EC G3HT39WW (1.13)

q60, 2021-09-26:
any way this works w/ my machine? i am interested in using `42T4235` battery.

leecher1337, 2021-12-02:
I'm unsure if the Embedded controller chip in chips.txt is really correct.
I have a L530 here, which should be the same hardware, and it actually contains a Nuvoton NPCE885G EC (also according to the schematics: http://laptop-schematics.com/view/8924/ ).
This would be a controller with CR16CPlus CPU (CompactRISC), which also matches the detected architecture of the EC code in the $01D4000.FL1  BIOS flash file.
Firmware starts at offset 0x4081D0 in BIOS image file starting with "IN", shortly followed by the Copyright string "Copyright 1996-1999, all rights reserved  Insyde Software Corp.".

There seems to be a thread a Thinkpad forum about the X210 EC, which seems to be the same model:
https://forum.thinkpads.com/viewtopic.php?p=837219#p837219 
Unfortunately, the GHIDRA-project associated with it is no longer available at the filehoster, would have been an interesting starting point for analysis.
There are 2 repositories with tools for EC patching and Checksum calculation for X210:

https://github.com/harrykipper/x210
https://github.com/jwise/x2100-ec

Maybe these informations are helpful for someone knowledgeable enough to patch the firmware.

leecher1337, 2021-12-03:
Some mor information: The BIOS Check for Battery is in Section_PE32_image_CC71B046-CF07-4DAE-AEAD-7046845BCD8A_LenovoVideoInitDxe.efi_body.bin 
Maybe useful for reversing to know command?
```
  if (LenovoEcServiceIf->Command(LenovoEcServiceIf, 0x88) & 1)
  {
    v3 = LenovoEcServiceIf->Command(LenovoEcServiceIf, 0xC2);
    if ( v3 < 0 && !(v3 & 0x20) )
    {
      ((void (__fastcall *)(EFI_SIMPLE_TEXT_OUTPUT_PROTOCOL *))EfiHandle->ConOut->ClearScreen)(EfiHandle->ConOut);
      sub_2850(15i64);
      sub_2864(0i64, 0i64);
      v5 = L"This system does not support batteries that are not genuine Lenovo-made or \n"
            "\rauthorized. The system will continue to boot, but may not charge unauthorized \n"
            "\rbatteries. \n"
            "\r\n"
            "ATTENTION: Lenovo has no responsibility for the performance or safety of \n"
            "\runauthorized batteries, and provides no warranties for failures or damage \n"
            "\rarising out of their use. \n"
            "\r\n"
            "Press the ESC key to continue.";
      if ( v4 & 0x50 )
        v5 = L"The battery installed is not supported by this system and will not charge.\n"
              "\rPlease replace the battery with the correct Lenovo battery for this system. \n"
              "\rPress the ESC key to continue.";
      sub_287C(v5);
```

leecher1337, 2021-12-03:
As far as I understand, X210 has a 7-row keyboard, so maybe some tables can be ported from X210 firmware?
The only table that I found that has slight differences is the one that gets used wihtin a subfunction of Translation function ( scancode_do_trans). Any idea what that could possibly be (left is x210, right is L530)?

[image]

leecher1337, 2021-12-04:
As I'm not experienced with Firmware Reverse engineering, Ghidra and 
embedded stuff, I was unable to find the battery functions
in the L530 firmware.
I tried to find some clues in the firmware in order to find out
where these battery authentication functions are, but they seem to
work differently compared to X210 firmware.
In X210 firmware, there are command packets for battery communication
(struct `BATT_REQ_QUERY`, see `bat_state_Queries` and `bat_prop_queries`)
that get sent via SMBUS. But in L530 firmware, I unfortunately 
cannot find such a mechanism.
There also isn't much documentation on how the keyboard scancodes
should map (to fix INS/DEL/POS1 etc. keys with 7-row keyboard).
I found some tables referenced in `scancode_do_trans`, but I don't
know how to map the values.
So as I don't have the expertise, I'm dumping my ghidra DB in the
hope that someone can pick it up and fix keyboard and battery.

Interestingly, there are 2 firmware images in the BIOS image:
1 starts at 0 (containing String `G3HT40WW(1.14)`) and one starting
at 0x8000 (NB: Load address 20100 !). I guess the second one is
the real one, even though, they seem somewhat similar.
Would be interesting what the first one is...?

In Ghidra project, there are various parts.
The Project is derived from https://github.com/jwise/x2100-ec/blob/master/notes/x2100-ec.gar by @jwise
and I added the 2 firmware images from the dump:

| File | Description |
|------|-------------|
| ec.bin     | This is the second Firmware from L530. This should be worked on. |
| fw-top.bin | This is the first  Firmware from L530. Don't know much about it. |
| x2100-ec.bin-patched | This is the x2100 firmware similar to L530, which I was comparing with |
| x2100-ec.bin | x2100 Working base, can be ignored |
| newec-gpe-patch.bin | New x2100 embedded controller FW? Not similar to L530 FW |

Hopefully, someone can find out the inner workings.
Project: [x2100-ec_2021_12_04.zip](https://github.com/hamishcoleman/thinkpad-ec/files/7654738/x2100-ec_2021_12_04.zip)
Firmware: [fw.zip](https://github.com/hamishcoleman/thinkpad-ec/files/7654737/fw.zip)

jwise, 2021-12-04:
Probably worth finding the EC command entry point and figuring out where it handles 0x88 and 0xC2.  If you can replicate those results using something like https://github.com/exander77/x2100-ec-sys then you can pretty easily figure out what those entry points read, and then patch out whatever checks generate them (and whatever else refers to that memory...).

leecher1337, 2021-12-05:
Updated Ghidra project, see `bat_authenticated` function

My guess is that battery authentication routine is right before main():

```
                             **************************************************************
                             *                          FUNCTION                          *
                             **************************************************************
                             long __ptrcall2 bat_authenticated(uint * param_1, uint *
             long              R1R0:4         <RETURN>
             uint *            R3R2:4         param_1
             uint *            R5R4:4         param_2
             undefined4        Stack[0x0]:4   param_1_00                              XREF[1]:     00031960(*)  
                             bat_authenticated                               XREF[1]:     FUN_0003161a:0003176e(c)  
        00031950 e7 01           push       $0x7, R7, ra
        00031952 bf 60 ea ff     ADDD       $0xffea,SP
        00031956 00 05 28 00     MOVD       $0x28,R1R0
        0003195a f0 61           ADDD       SP,R1R0
        0003195c 27 55           MOVD       param_1,R8R7
...
```

So maybe just patching it to:
```
        00031950 e7 01           push       $0x7, R7, ra
        00031952 10 5a           MOVW       $0x1,R0
        00031954 e7 03           popret     $0x7, R7, ra
```
works. Any volunteers? ;-)

leecher1337, 2021-12-05:
@jwise 
> Probably worth finding the EC command entry point and figuring out where it handles 0x88 and 0xC2. If you can replicate those results using something like https://github.com/exander77/x2100-ec-sys [...]

Seems that there are no 0xB... class commands in the firmware, so I guess I have to bite the bullet and flash it to the BIOS with flashrom and hope that I didn't mess up, I guess? Not sure if I want to risk it without any equipment for recovery.
Can you confirm that my patch is valid regarding patched instructions?

leecher1337, 2021-12-12:
# Patching L430/530 for a classic 7-row Keyboard

So I had a look at the Keyboard handling routines and found out where they keyboard layout table lies and how to edit it. The results are plenty tables that might become handy for you to patch your own keyboard layout.

## Location of the keyboard table
In Firmware G3HT40WW(1.14) there are 4 layout tables located at offset `0x39916`. They each have the following format (I don't count the prefix length byte in offset so that you have a "clean" keyboard layout offset matrix for the following tables):

## Table format

Offset | Length | Description
-------|----------|----------------------------------------
Prefix | 0x01   | length of table at offset 0xa2 in bytes
0x00   | 0x90   | Mapping table for the keys 0-144
0x90   | 0x12   | bitfield, specify whether to map a key via following Fn Mapping-table or not
0xa2   |        | Fn-Modifiers enabled in mask at offset 0x90, Array of 2 bytes per entry: [Keypress, Fn-Keypress]

Here is the default layout:
```
Prefix: 54
Offset      0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F

00000000   0E 05 0A 2E 36 55 16 4E  18 00 A2 A1 00 00 00 8C 
00000010   16 1E 26 25 3D 3E 46 45  1A 1E 1C A0 A3 00 00 00
00000020   15 1D 24 2D 3C 43 44 28  6A 00 04 00 00 00 00 00
00000030   0D 58 0C 2C 35 5B 14 54  66 82 02 00 00 00 88 00
00000040   1C 24 23 2B 3B 26 4B 4C  5D 00 00 9D 00 00 00 00
00000050   76 61 0E 34 33 12 64 52  10 00 08 00 A7 8A 00 00 
00000060   1A 22 21 2A 3A 41 49 5D  5A 00 06 A4 00 00 89 8D
00000070   67 8E 00 22 31 51 13 4A  20 A9 A8 A5 A6 8B 00 00
00000080   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
00000090   44 01 00 07 80 04 44 04  22 04 24 05 00 04 08 01
000000A0   00 00 C1 C1 10 B0 C3 C3  30 B1 C5 C5 30 B2 C7 C7
000000B0   10 B3 C9 C9 30 B4 06 CB  10 C2 04 CD 10 C3 0C CF
000000C0   10 C4 03 D1 10 C5 0B D3  10 C6 80 D5 10 C7 0A D7
000000D0   30 BB 01 D9 30 BC 09 DB  30 B9 78 DD 10 B8 07 DF
000000E0   30 BA 29 E1 10 B6 32 E3  10 D0 1B E5 10 D1 42 E7
000000F0   10 D2 4D E9 10 D3 00
```

Per default, a key in the table (assignment HW-scancode of 7-row Keyboard to keycode location in table will be shown in the following table) is roughly assigned to the specified scancode of Set 2. The EC contains the standard translation table 
to Set 1 and the CPU-Translation table. For details on the tables, see [here](https://www.win.tue.nl/~aeb/linux/kbd/scancodes-10.html). Thsi is pretty much standard stuff, EC will take care of Make and Break codes accordingly.
For codes >=80, there is a mapping table at offset 35f7c which contains of 2 bytes per entry. First byte is scancode and second byte is index in function pointer table at 34e6c which specifies on how to translate the table entry into a keycode. 
For you to create your keyboard layout, here is a list of some important "special" keycodes that are provided by this table:

Keycode	| Set 2 Scancode	| Description
-------------|-------------------------|------------------
80	| 83	| F7
81	| E0 5A	| Keypad Enter
82	| E0 1F	| Left GUI
83	| E0 27	| Right GUI
84	| E0 2F	| App
85	| E0 37	| Keyboard Power
86	| E0 3F	| System Sleep
87	| E0 5E	| System Wake
B8	| E0 3B	| Stop
B9	| E0 34	| Play/Pause
BA	| E0 23	| Mute
BB	| E0 32	| Volume up
BC	| E0 21	| Volume down
BD	| E0 48	| Mail
BE	| E0 10	| Scan prev. Track
BF	| E0 3A	| WWW Home
C0	| E0 38	| WWW Back
C1	| E0 30	| WWW Forward
C2	| E0 28	| WWW Stop
C3	| E0 20	| WWW Refresh
C4	| E0 18	| WWW Favorites
C5	| E0 2B	| Calculator
C6	| E0 40	| My Computer
C7	| E0 50	| Media select
[... 268 lines cut ...]
00000060   1A 22 21 2A 3A 41 49 5D  5A 00 06 C0 22 00 89 8D
00000070   67 8E 00 32 31 51 13 4A  29 1E 1C C1 1A 8B 00 00
00000080   00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
00000090   04 18 00 30 00 24 40 04  00 04 24 15 00 14 00 16
000000A0   00 00 C1 C1 10 B0 C3 C3  30 B1 C5 C5 30 B2 C7 C7
000000B0   10 B3 C9 C9 30 B4 06 CB  10 C3 E7 CD 10 BD 0C CF
000000C0   10 C4 03 D1 10 C5 0B D3  10 C6 80 D5 10 C7 A3 D7
000000D0   30 BB A2 D9 30 BC A6 DB  30 B9 A8 DD 10 B8 A9 DF
000000E0   30 BA A4 E1 10 B6 E9 E3  10 D0 9D E5 10 D1 A7 E8
000000F0   10 D2 10 B7 10 D3 00
```

You can try it out by compiling [https://github.com/exander77/x2100-ec-sys](https://github.com/exander77/x2100-ec-sys) and loading it with write support:
`modprobe x2100_ec_sys write_support=1`

Then verify that you have the correct firmware and the original table in place:
`dd if=/sys/kernel/debug/ec/ec0/ram bs=1 count=247 skip=$[0x366BD] 2>/dev/null | hexdump -C`

If it is the correct table from the top of this document, then you can patch the attached scancode.map in its place:

`dd if=scancode.map of=/sys/kernel/debug/ec/ec0/ram bs=1 skip=$[0x366BD]`

Keyboard map should be in place until you remove battery and power, to make it permanent, you need to reflash bios with patched EC firmware.

Have fun.

jwise, 2021-12-12:
Oh wow, wait, does the EC hotpatch stuff work out of the box without having to first patch the EC firmware?  That's crazy, if so.  And a serious security problem.

Anyway, this would be good news for you, since it means you can try hotpatching what you suspect to be the battery authentication routine without worrying about bricking your machine.

leecher1337, 2021-12-12:
Yes, hotpatching works out of the box :-)
Anyway, I do have a genuine battery, just wrong "model", so It's not "just" the auth check to patch, I shall have a look at the battery state machine later.

jwise, 2021-12-12:
Well, someone particularly daring could probably use this to write software for the EC that overwrites the flash descriptor, and therefore would allow you to `me_cleaner` the machine...

leecher1337, 2021-12-12:
When updating EC firmware, is this just a matter of Checksumming new firmware with checksum.c and placing it in the previously dumped BIOS firmware image with dd and flashing it back with flashrom or are there any other checksums to consider?

leecher1337, 2021-12-17:
I would like to ask if it is possible that someone with an programmer that
is capable to do recovery can test if updating the EC Firmware would cause problems 
possibly with some checksums, or if it is enough to just apply the patched EC 
firmware to the BIOS image and reflash via flashrom.
This would be very helpful. In order to carry out the test, I'd request you to
do a harmless patch of the EC keyboard map, which doesn't have an effect on
the proper operation of the keyboard, but shows if changing EC firmware would
cause some CRC error to go off.

The EC firmware is divided into 2 parts. The first part is the loader that 
loads attached real firmware image to address 20100.
As only the loader-Part of the firmware is checksummed, it is not necessary
needed to recalculate the checksum of the loader part.

So for testing, the following steps should be carried out.

1) Flash most recent stock firmware with EC Firmware version Firmware G3HT40WW(1.14)
2) Dump BIOS with:

   `flashrom -p internal -r current-bios.bin --ifd -i bios`

3) Check BIOS image so verify that we are patching the correct region

   `dd if=current-bios.bin bs=1 count=16 skip=$[0x41E5BD] 2>/dev/null | hexdump -C`

   This should result in the first row of the keyboard map:

   `00000000  0e 05 0a 2e 36 55 16 4e  18 00 a2 a1 00 00 00 8c  |....6U.N........|`

4) Patch an empty key mapping at offset C for test purposes:

   ```
   cp current-bios.bin new-bios.bin
   echo -ne "\x18" | dd conv=notrunc of=new-bios.bin bs=1 seek=$[0x41E5C9]
   ```

5) Flash back BIOS to machine

   `flashrom -p internal -w new-bios.bin --ifd -i bios`

6) Shutdown machine, remove power connection and battery for at least 30 seconds so
   that embedded controller is forced to reinitialize

7) Power up machine, install and load https://github.com/exander77/x2100-ec-sys

   `modprobe x2100_ec_sys`

8) Check memory region if the byte at offset C has changed:

   `dd if=/sys/kernel/debug/ec/ec0/ram bs=1 count=16 skip=$[0x366BD] 2>/dev/null | hexdump -C`

    Output should now read:

   `00000000  0e 05 0a 2e 36 55 16 4e  18 00 a2 a1 18 00 00 8c  |....6U.N........|`

I would be very grateful if someone can check this so that I can permanently
install the 7-row keyboard firmware fix then.
Thank you very much in advance!

unrelentingtech, 2021-12-17:
I can test this! Can you give me a test patch that swaps some keys around (e.g. Z and X) for a visible effect rather than having to check the memory?

> write software for the EC that overwrites the flash descriptor, and therefore would allow you to me_cleaner the machine

No need for that when [1vyrain](https://github.com/n4ru/1vyrain) exists. (Note: DO NOT use disable bits in me_cleaner like `-S`, that fails to boot, only do the cleaning)

leecher1337, 2021-12-17:
@unrelentingtech Thank you for your help!
Sure, we can change something significant too, so to swap Z and X:

3. Check BIOS image so verify that we are patching the correct region

dd if=current-bios.bin bs=1 count=16 skip=$[0x41E61D] 2>/dev/null | hexdump -C

This should result in the first 6th row of the keyboard map:

`1a 22 21 2a 3a 41 49 5d  5a 00 06 a4 00 00 89 8d  |."!*:AI]Z.......|`

4. Swap Z and X at offset 60:

```
cp current-bios.bin new-bios.bin
echo -ne "\x22\x1a" | dd conv=notrunc of=new-bios.bin bs=1 seek=$[0x41E61D]
```

In case it doesn't load the new EC fw, maybe we need to change FW version in header, but I will give instructions for this just in case that it doesn't work that way.

unrelentingtech, 2021-12-18:
Yes, it worked! \o/

<details>
<summary>i don't like flashrom's region handling / padded format, here's how i patched the whole rom with ifdtool</summary>

```
% ifdtool -x ~/l430.updated.rom
% cp flashregion_1_bios.bin bios_swapped.bin
% echo -ne "\x22\x1a" | dd conv=notrunc of=bios_swapped.bin bs=1 seek=$[0x41E61D-0x400000]
% ifdtool -O l430.kbtest.rom -i bios:bios_swapped.bin ~/l430.updated.rom
```

</details>

BTW, I don't think there's even a way for it to "not load" anything. Looks like this EC does not have its own flash, it's all always loaded from the main SPI flash: first, flashing the stock 2.04 bios also reverts the EC version to 1.13 (as shown by the setup screen), and also there is no "flashing the EC" step after any update (just flashed from stock to 2.76 from the official bootable ISO, that seems equivalent to what my manual updates did).

unrelentingtech, 2021-12-18:
> you only did ifdtool splitting because of your personal preference

Yes, mostly because I do the modifications on my desktop, while flashrom is on a raspberry pi.

> Battery authentication is another interesting topic with this firmware

Indeed. Is there any info from other machines with a similar EC that have already been patched?

leecher1337, 2021-12-19:
Now took a closer look

`echo -ne "\x0c\x60\x10" | dd -of=/sys/kernel/debug/ec/ec0/ram bs=1 seek=$[0x1004a]`

Should temporarily enable the battery. Will try to make a patch so that it permanently accepts any battery tomorrow (will need to modify some code of the state machine)

leecher1337, 2021-12-19:
# Patching Thinkpad L430/L530 battery check

## The Problem
The xx30 Series of Thinkpads unfortunately incorporate some kind of battery
check that makes it impossible to use an aftermarket and/or xx20 series
battery, even though i.e. T420 battery will perfectly fit into L530 and
also works.
The reason is some kind of challenge/response battery athentication mechanism
in the embedded controller. This was discovered by Dr. Matthew Chapman for his
X230T Thinkpad and very well documented in [his blog](https://zmatt.net/unlocking-my-lenovo-laptop-part-2/).
Unfortunately, the Thinkpad L430 and L530 series use a different embededd
controller (Nuvoton NPCE885G), so work had to be done to also patch this type 
of controller.

## The authentication routine

The authentication routine for battery authentication as pretty similar to
the one that is described by ZMatt. Mainly just the order of steps is
different. 
In Firmware G3HT40WW(1.14) the state machine routine is at `288ac`.
The state of the state machine including various flags is a 3 byte array at 
`1004a`. The first 2 bytes can be seen as a little endian word, the 3rd 
as a byte. However, to avoid confusion, I will address each byte seperately
in my description, so "byte 1" is at `1004a`, "byte 2" is at `1004b`, etc.
The lower 5 bits of the first word are indicating the state of the 
autentication state machine which I partly will describe here, the other bits
are various flags.

State | Description
------|-------------------------
1     | Init
2     | Wait for challenge to be constructed. The state machine for challenge construction and validation is at `3161a`. Start write command 0x27 (with 17-byte challenge)
3     | check success, retry if necessary
4     | start read command 0x28
5     | check success, retry if necessary
6     | start write command 0x3c (with 4-byte challenge)
7     | check success, retry if necessary
8     | start read command 0x3c
9     | check success, retry if necessary. Challenge is validated at state machine in `3161a`. If it succeeds, bit 0x40 in byte 2 of our 3 byte array at `1004a` is set.
12    | validate battery response

There are more states, but they are not interesting to us for our purpose.
The 4byte challenge, that gets written in Step 6,  seems to be partly composed 
of current system date in bytes 3 and 4.
The system date can be set by BIOS with the following 91h commands:

Command | Description
--------|------------------------
5Dh xx  | Set current day
5Eh xx  | Set current month
5Fh xx  | Set current year

If there is a failure in communicating with the Battery via SMbus, 
there is a error recovery handler at `282c0` which resets the
state machine back to state 1 and saves the prvious state in
`1004a` byte 2.

If communication fails 5 times in a row, there is a failure handler 
at `286d4` that clears some bits of `1004a` byte 2 (described later):

The state changes are:
States   | Description
---------|-------------------------
2, 4     | Clear bit 6     in byte 2
6, 8     | Clear bit 5, 7  in byte 2
19,21,23 | Reset state machine to 0

In States 2-8, it sets bit 4 in `1004a` byte 3 to indicate failure.

`1004a` Byte 2 flags  (Bit numbers are zero-based, little endian):
[... 166 lines cut ...]
`dd if=current-bios.bin bs=1 count=16 skip=$((0x3e7f00+0x28cea)) 2>/dev/null | hexdump -C`

This should result in the following original code:

`60 7b 01 00 9c 11 72 5d  22 5f 20 55 20 4c 20 61  |`{....r]"_ U L a|
`

4) Apply patches to image

```
echo -ne "\x60\x73\x01\x00\x50\x73\x01\x00\xe0\x18\x2c\x00" | dd conv=notrunc of=current-bios.bin bs=1 seek=$((0x3e7f00+0x28cea))
echo -ne "\xe0\x18\x5a\x00" | dd conv=notrunc of=current-bios.bin bs=1 seek=$((0x3e7f00+0x28c68))
echo -ne "\xc0" | dd conv=notrunc of=current-bios.bin bs=1 seek=$((0x3e7f00+0x28ade))
echo -ne "\xe0\x18\x54\x00" | dd conv=notrunc of=current-bios.bin bs=1 seek=$((0x3e7f00+0x28a72))
```

5) Flash back BIOS to machine

   `flashrom -p internal -l layout -w new-bios.bin -i ec`

6) Power down machine, force reload of EC firmware by removing battery and AC power for 30 seconds

This has not been tested yet, but should work.

Good luck!

jwise, 2021-12-20:
You need to recompute the EC checksum before you flash it. Otherwise the machine won't boot.

Dynamically patching possibly-live code can be tricky. In my x2100 EC patch set I have some tools to build patches and to build a "safe" patch loader that operates atomically. If I remember in the morning I'll try to write more about it...

leecher1337, 2021-12-20:
> You need to recompute the EC checksum before you flash it. Otherwise the machine won't boot.

No, I already explained that:

The EC firmware is divided into 2 parts. The first part is the loader that
loads attached real firmware image to address 20100.
As only the loader-Part of the firmware is checksummed, it is not necessary
needed to recalculate the checksum of the loader part. 
It will remain the same, when just changing the data that gets loaded.
That was the whole point of the test to check whether there is some other 
hidden checksum in it, but as @unrelentingtech was able to patch it without
issues, there doesn't seem to be a second check.

> Dynamically patching possibly-live code can be tricky. In my x2100 EC patch set I have some tools to build patches and to build a "safe" patch loader that operates atomically. If I remember in the morning I'll try to write more about it...

I just wrote my own using just bash, dd and hexdump that verifies old contents and allows hot-patching and image patching of files and sequences, I will upload it to a seperate repo soon so that you can also check, if I missed something.

leecher1337, 2021-12-23:
@unrelentingtech 
How did you flash the firmware on the L530?
I tried `flashrom -p internal -l layout -w new-bios.bin -i ec`
and this resulted in:

```
Found Winbond flash chip "W25Q64.V" (8192 kB, SPI), mapped at physical address 0x00000000ff800000.
Enabling flash write... SPI Configuration is locked down.

Reading old flash chip contents... Transaction error!
FAILED
```

Maybe this is due to a different flash chip than yours, could it be...?

unrelentingtech, 2021-12-23:
> @unrelentingtech 
> How did you flash the firmware on the L530?

With a raspberry pi and a chip clip.

You can only do internal with the 1vyrain "jailbreak" (see that thread where I was summoned from :D) on early enough firmware (2.54 or lower)

leecher1337, 2021-12-23:
As 1vyrain FAQ states:
```
The mod is fully compatible with EC modifications, but it must be flashed before you use 1vyrain to update your BIOS. 
```
I guess I have to find out how thinkpad-ec modifies the .FL1 files and then patch these to flash with Lenovo FW upgrade program instead of flashrom? It looks to me as the modifications just need to be carried out in the .FL1 file without any additional steps, as EC Firmware is not encrypted on L430 models. I thought that there are some UEFI checksums or stuff like that to consider, but I only find call of mec-tools which are for the encrypted firmware and L430 firmware is not encrypted and doesn't even need to be checksummed, as checksum of loader remains the same.
So could it be that it's just patching .FL1 file (FW starts at offset 0x4001d0 in .FL1 file)? Would you be able to try it out?

unrelentingtech, 2021-12-24:
The "flashed before" thing is mostly meant for those other models that have internal flash on the EC, or whatever they have. In our case this only means that flashing the pre-baked firmware image that comes on the 1vyrain ISO (for now only on a test version for L430!) will overwrite the EC region.

But I'm not saying to use that image! You can just run the jailbreak and then use flashrom from the command line with your own stuff.

Maybe we can eventually get them to include the EC modifications in their image...

leecher1337, 2021-12-26:
@unrelentingtech Shit, I modified .FL1 file with patched EC Firmware and did:

`dosflash /sd /sn /ipf ec /file G3ET94WW\$01D4000.FL1`

to update EC firmware. This command is based on what the official Windows update program does: 
`winflash32 /silent /vcpu /sn /sd /v /exit /ipf bios ec /shutdown /file`

So I was under the assumption that it can detect EC firmware location within FL1 file, but seems that it didn't.
Any hints on how to recover? Seems I need to find someone with a chip clip and RaspPi...?

unrelentingtech, 2021-12-26:
The chip is.. not in the most convenient location. I've had to modify the bottom case for easy access:

[image]

Also my Pi reboots every time I connect the chip clip, but otherwise it's just standard external flashing.

leecher1337, 2021-12-26:
Thanks for the fast reply! Seems that it will take 1-2 weeks until I can visit someone with a chip clip to try it out.

However I'm very curious what went wrong with flashing. 
My assumption is that the dosflash utility flashed the start of the FL1 BIOS image to the EC area (as i.e.  on other models, it flashes FL2 files which directly only contain EC ROM), as only Power button lights up as soon as I plug in power cord.
But on the other hand, why does original flash utility call contain "bios" AND "ec" region as parameters, if it doesn't know the layout of the flash file. I also found code to validate EC images in the flasher that seems to only run for certain machine types (those with seperate FL2, supposedly), and which would warn if image if bogus.

If I understand currectly, the EC always loads its code from BIOS chip, so it's not possible that I bricked the EC, right?

For reproducability, here is what I did:
I put this .img onto a stick (`dd if=patched.l430.img of=/dev/sdX...`) and then modified autoexec.bat to also contain the /sn parameter (als FL1 states these informations are not used anymore and original Flasher update routine also uses this parameter):
[patched.l430.zip](https://github.com/hamishcoleman/thinkpad-ec/files/7777693/patched.l430.zip)
It is worth to mention that in BIOS USB Menu UEFI booting from removable media has to be turned on, even though Legacy Boot has to be used to boot the stick, otherwise, it won't boot.
If you dare, you can try it out and analyse what went wrong, but I don't know if there is a risk associated with it, even though you have an external flasher.

unrelentingtech, 2021-12-27:
On the L430 there is no EC flash, no "EC region" on the IFD level (EC FW is embedded in the `bios`) – so no flash tools should ever "know" about "it". Of course these vendor tools get reused between various models, so I'm not surprised that there is functionality that has undefined behavior on some models.

It has probably done something really stupid, I am curious now :) so I will run it soon and dump the result.

leecher1337, 2021-12-27:
@q60 In the meantime, until we analyzed the reason for BIOS Update failure, you can use the hot-patching feature to solve your battery problem. This should be pretty risk-free. I made a repository with a bootable ISO to apply the patches in memory:

https://github.com/leecher1337/thinkpad-Lx30-ec/releases/tag/20211227

unrelentingtech, 2021-12-27:
So here's what that did (on top of a clean 2.76 FW):

<details>
<summary>Relatively long diff of the whole SPI flash</summary>

```diff
@@ -11337,9 +11337,12 @@
 │00032c00│ 0e 1c 02 c8 f0 00 02 00 ┊ ff 40 00 0e 1c 03 c8 f0 │•••××0•0┊×@0•••××│
 │00032c10│ 00 2a 00 00 48 00 0e 1d ┊ 00 c8 f0 00 02 00 ff 40 │0*00H0••┊0××0•0×@│
 │00032c20│ 00 0e 1d 01 c8 f0 00 2a ┊ 00 00 48 00 0e 1d 02 c8 │0•••××0*┊00H0•••×│
-│00032c30│ f0 00 02 00 ff 40 00 0e ┊ 1d 03 dc f0 00 2a 00 00 │×0•0×@0•┊••××0*00│
-│00032c40│ 48 00 0e 1e 01 dc f0 00 ┊ 02 00 ff 40 00 0e 1e 02 │H0•••××0┊•0×@0•••│
-│00032c50│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
+│00032c30│ f0 00 02 00 ff 40 00 0e ┊ 1d 03 c8 f0 00 2a 00 00 │×0•0×@0•┊••××0*00│
+│00032c40│ 48 00 0e 1e 01 c8 f0 00 ┊ 02 00 ff 40 00 0e 1e 02 │H0•••××0┊•0×@0•••│
+│00032c50│ c8 f0 00 2a 00 00 48 00 ┊ 0e 1e 03 c8 f0 00 02 00 │××0*00H0┊•••××0•0│
+│00032c60│ ff 40 00 0e 1f 01 dc f0 ┊ 00 2a 00 00 48 00 0e 1f │×@0•••××┊0*00H0••│
+│00032c70│ 02 dc f0 00 02 00 ff 40 ┊ 00 0e 1f 03 ff ff ff ff │•××0•0×@┊0•••××××│
+│00032c80│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
 │*       │                         ┊                         │        ┊        │
 │00034000│ 0c 78 fc ff 00 00 00 00 ┊ ff ff ff ff ff ff ff ff │_x××0000┊××××××××│
 │00034010│ ff 1f 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │×•000000┊00000000│
@@ -13283,13 +13286,13 @@
 │0003c010│ ff 1f 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │×•000000┊00000000│
 │0003c020│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 10 00 00 00 │00000000┊0000•000│
 │0003c030│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │00000000┊00000000│
-│0003c040│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 f0 ff ff ff │00000000┊0000××××│
+│0003c040│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 ff │00000000┊0000000×│
 │0003c050│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
 │*       │                         ┊                         │        ┊        │
 │0003c090│ 00 01 02 03 04 05 06 07 ┊ 08 09 0a 0b 0c 0d 0e 0f │0•••••••┊•__•__••│
-│0003c0a0│ 10 11 12 13 14 15 16 17 ┊ 18 19 1a 1b 1c 1d 1e ff │••••••••┊•••••••×│
-│0003c0b0│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
-│*       │                         ┊                         │        ┊        │
+│0003c0a0│ 10 11 12 13 14 15 16 17 ┊ 18 19 1a 1b 1c 1d 1e 1f │••••••••┊••••••••│
+│0003c0b0│ 20 ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │ ×××××××┊××××××××│
+│0003c0c0│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
 │0003c0d0│ 80 1f 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │×•000000┊00000000│
 │0003c0e0│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 ff │00000000┊0000000×│
 │0003c0f0│ a1 10 0e 01 00 ae ee b8 ┊ 0f 38 01 00 00 00 00 00 │×•••0×××┊•8•00000│
@@ -13747,8 +13750,27 @@
 │0003dea0│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │00000000┊00000000│
 │*       │                         ┊                         │        ┊        │
 │0003ded0│ 00 00 ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │00××××××┊××××××××│
-│0003dee0│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
+│0003dee0│ a3 20 0e 1f 00 00 01 02 ┊ 03 04 05 06 07 08 09 0a │× ••00••┊••••••__│
+│0003def0│ 0b 0c 0d 0e 0f 06 06 06 ┊ 06 06 06 00 00 00 00 01 │•__•••••┊•••0000•│
+│0003df00│ 80 2f 01 04 04 01 01 00 ┊ 00 00 00 00 00 00 00 00 │×/•••••0┊00000000│
+│0003df10│ 00 00 00 00 00 00 00 00 ┊ 00 04 01 01 10 00 01 00 │00000000┊0••••0•0│
+│0003df20│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 ff │00000000┊0000000×│
+│0003df30│ 81 42 36 5e 27 00 68 00 ┊ 00 00 00 00 00 00 00 00 │×B6^'0h0┊00000000│
+│0003df40│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │00000000┊00000000│
 │*       │                         ┊                         │        ┊        │
+│0003df70│ 00 00 ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │00××××××┊××××××××│
+│0003df80│ 82 4a 00 01 02 03 04 05 ┊ 06 07 08 09 0a 0b 0c 0d │×J0•••••┊•••__•__│
+│0003df90│ 0e 0f 06 06 06 06 06 06 ┊ 00 00 00 00 00 00 00 00 │••••••••┊00000000│
+│0003dfa0│ 01 01 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │••000000┊00000000│
+│0003dfb0│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │00000000┊00000000│
+│0003dfc0│ 00 00 00 00 00 00 00 00 ┊ 00 00 ff ff ff ff ff ff │00000000┊00××××××│
+│0003dfd0│ a3 30 0e 20 00 36 5e 27 ┊ 00 d0 00 00 00 00 00 00 │×0• 06^'┊0×000000│
+│0003dfe0│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │00000000┊00000000│
+│*       │                         ┊                         │        ┊        │
+│0003e000│ 80 17 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │×•000000┊00000000│
+│0003e010│ 00 00 00 00 00 00 00 ff ┊ ff ff ff ff ff ff ff ff │0000000×┊××××××××│
+│0003e020│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
+│*       │                         ┊                         │        ┊        │
 │00040000│ ff 78 ff ff 00 00 00 00 ┊ ff ff ff ff ff ff ff ff │×x××0000┊××××××××│
 │00040010│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
 │*       │                         ┊                         │        ┊        │
@@ -81597,13 +81619,13 @@
[... 448 lines cut ...]
+│006f6090│ 68 00 49 00 6e 00 50 00 ┊ 72 00 6f 00 63 00 65 00 │h0I0n0P0┊r0o0c0e0│
+│006f60a0│ 73 00 73 00 00 00 e3 23 ┊ 10 00 47 00 7f 00 07 00 │s0s000×#┊•0G0•0•0│
+│006f60b0│ 00 00 00 00 00 00 e3 37 ┊ 10 00 32 00 59 00 07 00 │000000×7┊•020Y0•0│
[... cut ...]
+│006f6150│ 04 00 ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │•0××××××┊××××××××│
+│006f6160│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
 │*       │                         ┊                         │        ┊        │
 │00710000│ 5f 46 4c 41 53 48 5f 4d ┊ 41 50 15 00 00 00 00 00 │_FLASH_M┊AP•00000│
 │00710010│ 00 00 00 00 00 00 00 00 ┊ 00 00 00 00 00 00 00 00 │00000000┊00000000│

```

</details>

…yeah, look at these changes below 0x3fffff — it has corrupted the ME region!!

This is why I avoid vendor tools as much as possible.

leecher1337, 2021-12-27:
Wow, thanks you very much for trying! That is really weird, I suspected that EC code got trashed, but in fact, EC patch looks fine to me.
The BIOS version that you were flashing from my Image was 2.54 (G3ET94WW) because it was the BIOS that was in my target machine. I should have told you that, because in theory, maybe the EC program argument was ignored and it flahed the old Firmware as a whole and what we are seeing here are the diffs between the versions. I just didn't expect a change in another area than EC code. But when comparing the FL1 files of both versions (2.76 and 2.54), I saw more diff than in your dump. But these FL1 files are only UEFI-capsules, so I guess I  cannot do a straight compare and find the byte patterns in question.
The possibility that comes to my mind is, that - at least in theory - it could still be the EC patch, because when you tested the modification of the keyboard table, this change only modified a data table in the file, whereas the battery patch also modifies code and maybe I missed some CRC routine or something that checks code integrity (but not data)?

So in order to check if it's the flash utility to blame or if it's the patched EC code, on restoring Firmware, you could leave out the EC area from flashing and check if machine then comes up again or if it remains broken (until flash area gets restored), which would mean back to reverse engineering for me and find the integrity check routine.

unrelentingtech, 2021-12-27:
There is no way it flashed the whole firmware — the whole "update" happened *way* too quickly, the log output said "Total blocks of the image = 32".

I'll do more experiments tomorrow, but to determine if it's the battery patch, can you provide it in the same `dd`-command format as the key test patch? (or as an UEFIPatch)

leecher1337, 2021-12-27:
You can apply it on any file with any offset you like conveniently with this script:
https://github.com/leecher1337/thinkpad-Lx30-ec/tree/main/fwpat

It basically does all the dd-ing, you just need to call the "applypat" function with the desired offset and filename, see i.e.:

https://github.com/leecher1337/thinkpad-Lx30-ec/blob/1db880a5f5ee26130dc0a698467a303b67861ebe/fwpat/patchui.sh#L130

patches are the patches to apply, i.e.: `"bat kb"` would be keyboard and battery, but "bat" would be enough for our test case.
Is this OK for you?

unrelentingtech, 2021-12-27:
Yes. Initially when I viewed the battery file over there I saw a `/sys/kernel/debug` path and got scared away :D but yes, the scripts actually make sense.

leecher1337, 2021-12-27:
In case the patch is responsible for EC malfunction, there still could be 2 possibilities:

1) Some hidden function for code integrity check that I cannot find
2) Code patches cause a hang of the EC when called during initialization but work when patching after EC has initialized (as I verified patches do work by writing to EC RAM).

So in order to find this out, here is an additional test case by doing a harmless patch to the code to check if there is some kind of code integrity check that causes the malfunction:
Only change ` 10 50` at offset `31780` to `00 50`. This is code that never gets reached with the wrong battery and with a valid battery, it would mark it non-genuine, so just inverses the meaning of batt challenge-response check.
If this little change of a compare instruction still causes the EC to malfunction, I guess it must be some kind of hidden code integrity check.

unrelentingtech, 2021-12-28:
Today's discoveries:

- Seems like the EC does retain firmware in RAM and it's only reloaded if necessary. When externally flashing it is necessary to unplug the battery and wait a couple seconds, otherwise the changes won't apply (e.g. a broken EC fimware would still appear to "work" because it hasn't loaded)
- 2.76 with `00 50` does not boot (might be the wrong `10 50` haha, you didn't specify the base for the offset and whether this is a hex or dec number, so `0x3e7f00 + 0x31780` was the only interpretation that did lead me to a `10 50`)
- 2.54 with `bash patch.sh $((0x3e7f00)) l430.254.rom` works fine
- 2.76 with `bash patch.sh $((0x8000-0x20100)) l430.276.bios` + [FW patches](https://github.com/digmorepaka/thinkpad-firmware-patches/pull/13) + packed with ifdtool works fine

To confirm, here's what the changes were

<details>
<summary>the bat patch diff</summary>

```diff
@@ -81597,13 +81597,13 @@
 │00410940│ 00 18 1e 07 70 5d 00 5f ┊ 04 55 24 4c 04 61 34 4c │0•••p]0_┊•U$L•a4L│
 │00410950│ 24 00 01 00 4a 00 14 90 ┊ 81 49 b1 22 1f 00 04 90 │$0•0J0•×┊×I×"•0•×│
 │00410960│ b0 22 e0 ff 10 27 04 d0 ┊ e0 18 f6 06 00 c0 a2 8b │×"××•'•×┊×•×•0×××│
-│00410970│ 00 50 00 18 ec 06 70 5d ┊ 00 5f 02 55 22 4c 02 61 │0P0•×•p]┊0_•U"L•a│
+│00410970│ 00 50 e0 18 54 00 70 5d ┊ 00 5f 02 55 22 4c 02 61 │0P×•T0p]┊0_•U"L•a│
 │00410980│ 32 4c 22 00 01 00 4e 00 ┊ 00 c0 84 8e 72 59 ff c0 │2L"0•0N0┊0×××rY××│
 │00410990│ 15 fe 00 01 10 58 00 01 ┊ 72 5d 22 5f 20 55 20 4c │•×0••X0•┊r]"_ U L│
 │004109a0│ 20 61 30 4c 20 00 01 00 ┊ 4e 00 10 01 b5 58 11 00 │ a0L 0•0┊N0••×X•0│
 │004109b0│ b4 58 27 00 b3 58 16 00 ┊ 72 59 ff c0 cd 96 8f 60 │×X'0×X•0┊rY×××××`│
 │004109c0│ 00 50 00 18 9c 06 70 5d ┊ 00 5f 02 55 22 4c 02 61 │0P0•×•p]┊0_•U"L•a│
-│004109d0│ 32 4c 22 00 01 00 4a 00 ┊ 02 90 b0 22 e0 ff 30 26 │2L"0•0J0┊•××"××0&│
+│004109d0│ 32 4c 22 00 01 00 4a 00 ┊ 02 90 b0 22 e0 ff c0 26 │2L"0•0J0┊•××"×××&│
 │004109e0│ e0 18 7c 06 70 5d 00 5f ┊ 02 55 22 4c 02 61 32 4c │×•|•p]0_┊•U"L•a2L│
 │004109f0│ 22 00 01 00 4a 00 b0 58 ┊ 20 00 02 f3 1e 01 72 59 │"0•0J0×X┊ 0•×••rY│
 │00410a00│ ff c0 a3 fd 0f f0 72 5d ┊ 22 5f 20 55 20 4c 20 61 │××××•×r]┊"_ U L a│
@@ -81628,7 +81628,7 @@
 │00410b30│ 30 05 70 5d 00 5f 02 55 ┊ 22 4c 02 61 32 4c 22 00 │0•p]0_•U┊"L•a2L"0│
 │00410b40│ 01 00 4a 00 02 90 b0 22 ┊ e0 ff b0 26 09 00 e0 18 │•0J0•××"┊×××&_0×•│
 │00410b50│ 0e 05 72 5d 22 5f 20 55 ┊ 20 4c 20 61 30 4c 20 00 │••r]"_ U┊ L a0L 0│
-│00410b60│ 01 00 4a 00 30 7b 02 00 ┊ 80 18 f6 04 20 55 22 4c │•0J00{•0┊×•×• U"L│
+│00410b60│ 01 00 4a 00 30 7b 02 00 ┊ e0 18 5a 00 20 55 22 4c │•0J00{•0┊×•Z0 U"L│
 │00410b70│ 02 61 32 4c 22 00 01 00 ┊ 6e 00 02 b1 00 50 12 12 │•a2L"0•0┊n0•×0P••│
 │00410b80│ 72 59 ff c0 cd f3 00 50 ┊ 1d 11 70 5d 00 5f 02 55 │rY××××0P┊••p]0_•U│
 │00410b90│ 22 4c 02 61 32 4c 22 00 ┊ 01 00 4a 00 02 90 b0 22 │"L•a2L"0┊•0J0•××"│
@@ -81636,8 +81636,8 @@
 │00410bb0│ 34 4c 24 00 01 00 6e 00 ┊ 04 b1 10 30 04 f1 e0 18 │4L$0•0n0┊•×•0•××•│
 │00410bc0│ a0 04 72 5d 22 5f 20 55 ┊ 20 4c 20 61 30 4c 20 00 │×•r]"_ U┊ L a0L 0│
 │00410bd0│ 01 00 6e 00 00 83 01 00 ┊ 72 5d 22 5f 20 55 20 4c │•0n00×•0┊r]"_ U L│
-│00410be0│ 20 61 30 4c 20 00 01 00 ┊ 4a 00 60 7b 01 00 9c 11 │ a0L 0•0┊J0`{•0×•│
-│00410bf0│ 72 5d 22 5f 20 55 20 4c ┊ 20 61 30 4c 20 00 01 00 │r]"_ U L┊ a0L 0•0│
+│00410be0│ 20 61 30 4c 20 00 01 00 ┊ 4a 00 60 73 01 00 50 73 │ a0L 0•0┊J0`s•0Ps│
+│00410bf0│ 01 00 e0 18 2c 00 20 4c ┊ 20 61 30 4c 20 00 01 00 │•0×•,0 L┊ a0L 0•0│
 │00410c00│ 4a 00 50 7b 01 00 8c 10 ┊ 20 55 22 4c 02 61 32 4c │J0P{•0×•┊ U"L•a2L│
 │00410c10│ 22 00 01 00 4a 00 02 90 ┊ 10 49 00 50 05 10 10 00 │"0•0J0•×┊•I0P•••0│
 │00410c20│ 00 71 84 04 e4 10 10 00 ┊ 00 b1 84 04 72 5d 22 5f │0q×•×••0┊0××•r]"_│
```
</details>

<details>
<summary>the `00 50` diff</summary>

```diff
@@ -83854,7 +83854,7 @@
 │00419650│ 9a 12 12 00 90 31 3a 0a ┊ e6 12 70 00 01 00 bc 09 │×••0×1:_┊×•p0•0×_│
 │00419660│ 10 01 74 00 01 00 28 0a ┊ 72 00 01 00 e0 09 00 c0 │••t0•0(_┊r0•0×_0×│
 │00419670│ e2 01 4f 60 12 00 20 f1 ┊ 4c 00 12 00 10 f1 4a 00 │×•O`•0 ×┊L0•0•×J0│
-│00419680│ 10 50 14 10 b1 26 00 40 ┊ e3 10 b1 22 ff bf 13 00 │•P••×&0@┊×•×"××•0│
+│00419680│ 00 50 14 10 b1 26 00 40 ┊ e3 10 b1 22 ff bf 13 00 │0P••×&0@┊×•×"××•0│
 │00419690│ 10 f1 4a 00 b2 22 f7 ff ┊ 13 00 20 f1 4c 00 12 00 │•×J0×"××┊•0 ×L0•0│
 │004196a0│ 00 31 3a 0a 1e 03 a7 01 ┊ 01 01 12 00 00 f1 74 06 │01:_••×•┊•••00×t•│
 │004196b0│ 13 00 00 f1 18 0a 12 00 ┊ 00 f1 76 06 13 00 00 f1 │•00×•_•0┊0×v••00×│
```
</details>

Bonus content:

[... 1844 lines cut ...]
+│00713e80│ 65 af af 90 25 ad 55 d6 ┊ 91 a5 e6 df 91 a0 89 ef │e×××%×U×┊××××××××│
+│00713e90│ 8f 32 4b b5 ee c6 d5 8e ┊ bb 13 8e 43 5d 3d 72 4e │×2K×××××┊×•×C]=rN│
+│00713ea0│ 3a 4f 26 a6 67 b0 28 56 ┊ cf 6a 1c ea 30 ee 08 61 │:O&×g×(V┊×j•×0×•a│
+│00713eb0│ 2d 7d 42 8c fa de d6 ee ┊ d5 3b 94 0c 28 61 df 4f │-}B×××××┊×;×_(a×O│
+│00713ec0│ 4f f0 fe 21 db a6 cd 78 ┊ 8a 0f 07 28 5f d5 dd b5 │O××!×××x┊×••(_×××│
+│00713ed0│ d7 93 72 98 c9 6e 65 88 ┊ f2 a5 b7 a9 c3 75 0e 65 │××r××ne×┊×××××u•e│
+│00713ee0│ 12 bc d5 32 d4 5d ce c2 ┊ d1 65 5e b9 6c 4e a8 00 │•××2×]××┊×e^×lN×0│
+│00713ef0│ 07 ba 28 78 30 8a 0c 70 ┊ b5 54 58 50 b5 22 23 3e │•×(x0×_p┊×TXP×"#>│
+│00713f00│ df 61 4f e0 91 ee 60 1b ┊ 47 84 72 f7 ea 69 a5 28 │×aO×××`•┊G×r××i×(│
+│00713f10│ ca 4f f5 3a b7 18 0e 3e ┊ bf 87 31 87 0a 10 d9 c5 │×O×:×••>┊××1×_•××│
+│00713f20│ 34 ab 00 7d 10 07 e9 ba ┊ c2 ed 41 a9 41 c0 bf ea │4×0}••××┊××A×A×××│
+│00713f30│ 9e 82 fb 54 9d 85 b9 81 ┊ 36 5f 01 f1 9c 1a b9 b8 │×××T××××┊6_•××•××│
[... cut ...]
+│00713f80│ ff ff ff ff ff ff ff ff ┊ ff ff ff ff ff ff ff ff │××××××××┊××××××××│
 │*       │                         ┊                         │        ┊        │
 │00740000│ ea d0 ff 00 f0 00 00 00 ┊ 00 00 00 00 00 00 27 2d │×××0×000┊000000'-│
 │00740010│ d9 54 93 7a 68 04 4a 44 ┊ 81 ce 0b f6 17 d8 90 df │×T×zh•JD┊××•×•×××│
```

</details>

P.S. the EC datasheet link in your readme seems to return a 0-byte file. and honestly I could not find a datasheet for this anywhere on the internet. If you have it, please upload it somewhere!

leecher1337, 2021-12-28:
Thank you for the extensive Tests!
As I bricked my ThinkPad, I was unable to validate my 2 Byte Patch, offset calculation looks correct.

Anyway, as patching the EC directly worked (glad to hear that, because I would have been unable to identify CRC Check function), I conclude that your first assumption about the FW flash tool messing up the ME region ist correct then? 
So options are:
1) Trying to flash patched FL1 with default Tool options to Check If it also corrupts ME
2) Unlock Chip with 1vyrain and Patch EC Region directly
3) Trying to find the bug in the flash tool that causes corruption (looks like a ton of fun ;-) )

Would you agreed with that? And would you bei able to test 1) ? Or did I misinterpret the results?

unrelentingtech, 2021-12-28:
Yes, I'll test that. Actually I'm not sure why you've added that option at all? What is that option even for?

leecher1337, 2021-12-28:
You mean the option to specify EC Region? Well, I just took the autoexec template from this project and patched FL1 and it generally makes sense to only patch EC region so that the patch is universal regardless of the installed BIOS version, as long as EC FW version matches. And it is faster than flashing whole BIOS. 
It is really surprising that flasher uses the correct region from FL1, but still corrupts other areas.

Regarding datasheet: I can download the PDF without issues.

unrelentingtech, 2021-12-28:
The `ec` arg was already there, but you told me to add the `/sn` flag. What does `/sn` do?

> I can download the PDF without issues

Right, worked fine from a different IP address. Seems like it's just a schematic though, no documentation, but still interesting.

leecher1337, 2021-12-28:
You can get all parameters with `dosflash /help`
`/sn`removes check for PartNumber.
If you try without it, you get an error, that PartNumber does not match. However, if you look at .FL1:

```
Offset      0  1  2  3  4  5  6  7   8  9  A  B  C  D  E  F

006C61D0   43 4D 44 42 0C 00 00 00  23 00 00 00 42 00 17 42   CMDB....#...B..B
006C61E0   09 05 02 11 42 1E 05 02  29 42 36 05 02 42 42 4F   ....B...)B6..BBO
006C61F0   05 02 57 42 69 6F 73 49  6E 66 6F 00 42 6F 61 72   ..WBiosInfo.Boar
006C6200   64 49 64 00 4E 6F 4C 6F  6E 67 65 72 55 73 65 64   dId.NoLongerUsed
006C6210   00 50 61 72 74 4E 75 6D  62 65 72 00 4E 6F 4C 6F   .PartNumber.NoLo
006C6220   6E 67 65 72 55 73 65 64  00 52 65 6C 65 61 73 65   ngerUsed.Release
006C6230   44 61 74 65 00 4E 6F 4C  6F 6E 67 65 72 55 73 65   Date.NoLongerUse
006C6240   64 00 56 65 72 73 69 6F  6E 00 4E 6F 4C 6F 6E 67   d.Version.NoLong
006C6250   65 72 55 73 65 64 00 FF  FF FF FF FF FF FF FF FF   erUsed.ÿÿÿÿÿÿÿÿÿ
```

And official call of WinFlash32 also contains this parameter which indicates that it must be ignored.
To test DOS flash the 'standard' way, just call `command` on prompt, it will start a menu that lets you update the FW.
You can also try to use the parameters for DosFlash, that I Specified above for winflash32 that I specified above, but there are actually 4 possible WinFlash32 calls in the .exe, so safest bet is to use menu driven `command`call first. (command.com on DOS disk actually isn't the normal command.com interpreter, but the Lenovo menu for flashing)

leecher1337, 2022-01-02:
@unrelentingtech I bought a cheap CH341A programmer and fixed it for 3.3V on data lines (seems the schematic has a design flaw and normally uses 5V instead of 3.3V on the data lines!) so that I can fix my broken machine.
Do I have to remove the CMOS battery before attaching the chip clip with the programmer to the flash chip?

unrelentingtech, 2022-01-02:
> Do I have to remove the CMOS battery before attaching the chip clip with the programmer to the flash chip?

I didn't remove *anything*, often not even the actual laptop battery lol

leecher1337, 2022-01-02:
Hm, crap, the CH431A works fine when attaching a Flash chip directly into the socket (did this to verify that it is working correctly after 3.3V fix), but when attaching the chip clip to the BIOS Chip on the L530, just the Orange RUN-Led lights up (no documentation can be found for this board to find out what this means) and the programmer isn't detected by the host system. I double-checked for the correct wiring, but cannot find a wiring error. Pin1 is connected through properly. 
Weird. So I guess my only chance is using the RaspPi then, hopefully I haven't ruined anything yet. Arrrgh

unrelentingtech, 2022-01-02:
> the programmer isn't detected by the host system

hm. Probably the same issue that causes the Pi to reset.

leecher1337, 2022-01-02:
As EC is also connected to SPI flash, maybe CH431A doesn't supply enough power to run both.
I also heard that EC jams the SPI flash if it can't find the firmware which makes flashing even more complicated without desoldering and resoldering the chip.

As it seems that it's not easy to fix my machine and carry out some more tests, if would still be interesting if flashing whole BIOS image including patched EC firmware with dosflash also causes these strange corruption problems.

Btw.: I guess the symtpoms that you were experiencing were the same than mine after dosing dosflash with just ec area: Machine just doesn't power on on pressing the power button and when plugging in PSU, Power led lights up but machine does nothing.

unrelentingtech, 2022-01-02:
Flashrom does identify it as a W25Qsomething, so probably not different.

> Machine just doesn't power on on pressing the power button

Yes, that's what happens with any kind of non-working firmware (e.g. me_cleaner disable bit too)

leecher1337, 2022-01-02:
Are you powering the Flash Chip (Pin 8 - VCC) via 3.3V Pin 1 of the RaspPi or are you using system battery/AC adapter to power the flash chip and just use the following pinout:

Pin Chip | Name | Pin RaspPi
--|---|----
1 | CS | 24
2 | MISO | 21
3 | not used | not used
4 | GND | 25
5 | MOSI | 19
6 | CLK | 23
7 | not used | not used
8 | 3.3V

unrelentingtech, 2022-01-02:
I do connect 3v3 to the Pi. I'm pretty sure the chip won't get any power from the system until the system is on (which requires the EC… so it must be able to turn enough power on without the firmware to be able to even read the firmware…?).

Just tested with a multimeter, no voltage on any pins when the laptop is off.

leecher1337, 2022-01-02:
Hm, weird.. After having no success with the CH431A, I tested chip clip with Raspberry Pi on another broken mainboard with the same flash chip in order to be on the safe side in case I brick something due to bad wiring or something. Triple-checked wiring (also checked clip cabling with pin headers), then put 3.3V on Pin 8, but nothing: Chip not detected. 
I'm running out of ideas now. 
Could be that the CH431A bricked both flash chips, but I doubt that when attaching a Flash Chip to the socket of the CH431A  programmer works fine without damaging the chip.

unrelentingtech, 2022-01-02:
> spispeed=32768

Whaaat?! That's ridiculously high, I've been using 1000.

Also, these clips like to not attach properly. In my case, if the Pi did not reset, it's not attached.

leecher1337, 2022-01-02:
Also doesn't work with 1000. Hmm, so you attach all wires to the clip, then attach 3.3V on pin8 and then the Raspberry Pi resets?
I just get a message about undervoltage for a short moment, may be due to my crappy power supply of the Pi, but as it's just a short undervoltage, I don't think it's the reason. Anyway, will try with a new Power Supply for the Pi tomorrow.

unrelentingtech, 2022-01-02:
I keep the clip always connected to the Pi, including 3v3 (using pin 17 for 3v3 since it's close to the others), when I need to flash I just connect the clip to the chip, if the attachment is good the Pi resets.

In-system programming is tricky, electricity is weird magic, this exact same Pi setup does *not* detect the flash chip in a different device I've disassembled recently (an IP camera) :/

leecher1337, 2022-01-02:
I found some schematics where pin 3 and 7 also need to be connected to 3.3V. Is my PIN layout above correct?

unrelentingtech, 2022-01-02:
Yes, I have these two unconnected.

leecher1337, 2022-01-02:
Verified that the chip clip is attached correctly with Multimeter probes, no luck so far.
But I don't really understand why it works for you with Pin 7 being LO (as not attached to VCC):

HOLD - this is a 'wait' pin for the SPI bus. When pulled low, it puts the SPI bus on hold. This is different than the CS pin because it doesnt stop the current transaction.

unrelentingtech, 2022-01-03:
Boards [usually do tie these pins to VCC](https://electronics.stackexchange.com/a/416258) (if they don't have fancy digital security control over WP), have you found something on the boardview suggesting otherwise?

leecher1337, 2022-01-03:
You are right, this sounds reasonable. As I read that the CH431A may connect these lines to VCC (which would be OK for flashing a chip in the Socket, which worked with this programmer), could it be that I trashed my board/chip by using the CH431A? :-/

jwise, 2022-01-03:
FYI -- I have heard noises that the Nuvoton ECs get upset if they fail to find a firmware image (in particular, they may continuously try to reread the flash, which woul dprevent ch341a from talking to a flash that is controlled by a sad EC).  I have heard from mjg59 that a board in such a state can be recovered by using an external power supply to power up Vcc on the flash and undervolting it sufficiently that the flash will operate but the EC will not boot.  I have not tried this myself on my x2100.

[39 more replies]