hamishcoleman/thinkpad-ec #204: Battery patch not working on x230

nathanstampfli, 2021-10-09:
I've been trying to make the battery patch work on my Thinkpad x230. I've installed the firmware version (2.75 and 1.14).
I used the command to enable the battery patch and disable the keyboard patch.
No problem booting on the usb device, I get the message to flash the Embedded controller firmware, I press any key and it does flash. After that the laptop reboots and shows "Flashing Embedded Controller.... Please do not power off" and then reboots again by itself.

Unfortunately, when I go into Windows, I still can not charge my x220 battery. 
I flashed the laptop several times, with an original x230 battery inside, without any battery, using different power adapters and still no luck getting the battery charged.

Any suggestions to what I could do differently?

DWC147, 2022-02-04:
Same with me, X230. I tried all the combinations of the battery and keyboard commands in various orders to see if I could get the battery issue fixed but to no avail.
If I set the battery option to "make patch_enable_battery clean", it would show just at the end of compiling that it was set, but if I used the "make patch_disable_battery clean" it didn't show any battery changes after compiling but did show the Keyboard changes which isn't the problem I'm fixing.
In any event, it did not switch off the battery validation.
I have flashed the BIOS to several different versions, and did use the 2.75  and EC 1.4.
I think that its impossible to enable the battery patch. Perhaps it could be enabled by default and then it might work.
Thanks for reading.

Xanth0k1d, 2022-02-09:
My suggestion is to `make clean`
then `make patch_enable_battery clean`
then compile the img.

Alternatively you can try to clone into a fresh folder and attempt again.

pavelselivanov, 2022-02-11:
>My suggestion is to make clean
>then make patch_enable_battery 

Thanks! That worked for me! Battery patch wasn't enable by default.