hamishcoleman/thinkpad-ec #208: not a new issue but still need help

DWC147, 2022-02-04:
I posted my info in someone else's similar report so I am posting here in its own space. Sorry for doubling up. (new to this);

Thinkpad X230. I tried all the combinations of the battery and keyboard commands in various orders to see if I could get the battery issue fixed but to no avail.
If I set the battery option to "make patch_enable_battery clean", it would show just at the end of compiling that it was set, but if I used the "make patch_disable_battery clean" it didn't show any battery changes after compiling but did show the Keyboard changes which isn't the problem I'm fixing.
In any event, it did not switch off the battery validation.
I have flashed the BIOS to several different versions, and did use the 2.75 and EC 1.4.
I think that its impossible to enable the battery patch. Perhaps it could be enabled by default and then it might work.
Thanks for reading.

Xanth0k1d, 2022-02-09:
My suggestion is to do a fresh clone, then
"The selected commands from the above list simply need to be typed in exactly as shown, at the command prompt, **before building the patched image**, while following the Step-by-step instructions (This is step 6)"
or maybe make clean with your old clone?

DWC147, 2022-02-10:
Hi, I tried the Clean command 1st, went thru all commands in order but it still fails to fix the battery issue.
I have included a screenshot of the last bit. maybe there's s glaringly wrong.
I'm not Linux user but have had reasonable experience with scripts in Windows.
Thanks in advance. Darrell

Xanth0k1d, 2022-02-10:
Honestly it shouldn't matter, perhaps you can just compile it in vbox. I can try to compile an image for you, but do back up your current 8mb chip with ch341a or something.

DWC147, 2022-02-10:
Ok, that would be really appreciated.I'll see if I can do a backup of my 8 mb chip which I presume is my BIOS.Thanks in advance. Darrell

Xanth0k1d, 2022-02-10:
The 4mb is BIOS
8mb is the firmware for the embedded controller aka EC.
4mb corresponds to the .fl1 file in the thinkpad bios update cd and 8mb corresponds to the .fl2 file.
You will find that unlike .fl1, which is roughly 4mb, .fl2 might be no where close to 8mb, because it's something like can incremental update( which makes it inconvenient in my opinion) this repo modifies the some parts of the 8mb only.

To back up you can try flashrom to read the internal flash(normally it is used to read the 4mb bios)

Or you can just use a ch341a with flashrom to clip the chip under the palmrest.

DWC147, 2022-02-10:
Hi, after watching a video on ch341a I think it's not something I can do as I would need to buy the unit and the technical side of it is beyond my experience.
I've had a look at flashrom and that might be the better way forward for me although even that is somewhat challenging.
I have no real Hardware experience at this level.
I have already reset the BIOS and the EC about a dozen times on this Laptop and spent more than 10 hours grappling with Linux, compiling commands and the process, so I believe that if your pre-compiled version doesn't work I'm sunk anyway.
I know it would be prudent to back it up but I think I will just go with your update.
I know I would be fully responsible for attempting a fix without a backup.

What do you reckon?
Cheers, Darrell

Xanth0k1d, 2022-02-10:
It is more of a privacy concern, what if I put something in the file that I gave you?
What I'm trying to do now is to give you a fixed guide on how to make it work. The image is just to test it out.

Xanth0k1d, 2022-02-10:
[patched.x230.zip](https://github.com/hamishcoleman/thinkpad-ec/files/8038499/patched.x230.zip)
[
[image]
What you can try to do now is:
Update your bios in windows, download the file from the lenovo website, matching g2uj31us.
Make sure the EC get's flashed as well (When you reboot, you should see the prompt of flashing EC)
Then reboot again and use your windows for a bit, then dd the img and flash the one I compiled for you.
I'd also appreciate if you can confirm that EC will start the flash even with a non genuine battery attached (Currently I'm waiting for a non genuine battery for my x230t)
](url)

DWC147, 2022-02-10:
I'm happy to test the not working battery although it is a genuine battery I purchased from Lenovo 3 weeks ago. I originally bought a non genuine battery but it wasn't recognised and thought at the time I needed a genuine one and wrote off the non-genuine battery although I still have it.
I will try both and let you know.

So do I use your patched version within my Linux Live usb enviroment first?
I presume I would come in on maybe 'step 8'?
The laptop is currently on Bios 2.75 (G2ETB5WW) and EC 1.14 (G2HT35WW).<https://outlook.live.com/mail/0/inbox/id/url> g2uj31us

I'm afraid I don't know what this means " then dd the img and flash the one I compiled for you.<https://outlook.live.com/mail/0/inbox/id/url>"

patched.x230.zip<https://github.com/hamishcoleman/thinkpad-ec/files/8038499/patched.x230.zip>

[x230darrel]

Xanth0k1d, 2022-02-10:
Well, you need to update/reflash your BIOS via windows first to be safe(get the bios update exe from lenovo) make sure you match the version.

Then proceed from step 8 via live usb

DWC147, 2022-02-10:
Ok, will do.

DWC147, 2022-02-10:
Bios update won't update to the version we want because the exe says its already at that version and then quits.
I will go with the Linux option step 8.

Xanth0k1d, 2022-02-10:
Aight M8
Fire up your live usb:
Go into the folder where you placed my patched .img

sudo dd if="my pacthed img" of=/dev/sdx
where the /dev/sdx is your usb drive you planning to boot the EC update.

the pactched image that I compile doesnt require you to use the make command, if you use that you are just generating another .img as you have figured out. dd is available as a system-wide tool, so you dont have to go into a particular folder to use it.

after you dd the usb drive, when you plug it on to windows you should be seeing a bunch of files, otherwise it means that I messed up on my end and gave you a corrupted .img file.

DWC147, 2022-02-10:
Thanks Hamish. I am just doing some personal errands and will try again a little bit later on.I'll let you know how it goes.Thankyou. Darrell.

DWC147, 2022-02-10:
No success I'm afraid.
I did that much quicker method of DD using your IMG. but it still doesn't fix the battery problem.
I had no issues during the process and so I thought I would just give you some info on the results.
I have attached some pictures that are relevant.
I don't expect you to fault find any more so I thank you greatly for your help so far.
When you get your battery, you might have the same issues and find a solution, let me know if you do.

Cheers and good luck, Darrell

Xanth0k1d, 2022-02-11:
Very interesting, I cant see your images. Possible for you to attach them on github?
Lets try to confirm somethings here:
1. Battery validate patch is applied (Shown in the log) - by applying this patch, battery validation should be disabled right? @hamishcoleman 
2. Did your thinkpad actually flash the EC? (After reboot, it shows a thinkpad logo, below which the text "flashing EC"

DWC147, 2022-02-11:
Hi Hamish, yes on both points, 1 & 2.There were photos of those steps in a zipped up file, but they didn't get to you. Including the log file.I have looked everywhere for my log in to GitHub but can't find it.I'm not able to log in but would love to get you those photos.I'll try again tomorrow morning and see if I can get into GitHub, if not I'll let you know and maybe I can upload them somewhere for you.Cheers, Darrell.

Battery validate patch is applied (Shown in the log) - by applying this patch, battery validation should be disabled right? @hamishcolemanDid your thinkpad actually flash the EC? (After reboot, it shows a thinkpad logo, below which the text "flashing EC"

DWC147, 2022-02-11:
Here are the files you requested
[Results.zip](https://github.com/hamishcoleman/thinkpad-ec/files/8052190/Results.zip)
.

DWC147, 2022-02-11:
Hi, I have uploaded the files into Github. Cheers. Darrell

closed 2022-02-11 by DWC147

Xanth0k1d, 2022-02-12:
Say, could it be a software issue now? Boot linux and see if your battery is charging.

DWC147, 2022-02-13:
Hi, I booted up on the USB-linux and the battery icon shows as red and when I click on it says 'Charging' 0%.
After 12 hours of charging there was no change to the % and when I unplugged the charger the laptop immediately shutdown.
I think there must be some other issue even though the very original battery does charge up and holds power for about half an hour slowly (quickly) discharging normally.
Thanks for all your help and good luck with your X230. 👍
Darrell.

Xanth0k1d, 2022-02-14:
You should re-open this issue, the patch may not be working for some x230s