hamishcoleman/thinkpad-ec #209: Flashing EC's Conditions

Xanth0k1d, 2022-02-08:
Hi, I got a ThinkPad x230t with EC version 1.13.
(I mean I could modify the script to download an iso with 1.13 ec but I'd like the 1.14 one because I'm slightly OCD)
I'm trying to update it to 1.14.
Currently just loading a DOS usb in legacy mode with dosflash to do so.
I had no issues flashing the BIOS, but can't seem to flash the EC (nothing after rebooting, doesn't show flashing EC)
Here are some of my concerns:
1. I modified the charger port to type c, and I hardwired the 10k resistor on the signal pin so the laptop will always think that a 65w charger is plugged in.
2. I do not have a genuine battery and I'm only ordering a replacement (reason why I'm attempting to use the ec mod just in case)
Hence my questions for discussion:
As mentioned, could it be that EC only updates if it detects a battery attached? And must it be a genuine ThinkPad battery?
Anyone has experiences successfully flashing EC without a battery/with a non genuine battery attached/with a 65w charger?
(Feeling stupid now because I cheaped out on the genuine battery for a 10 dollar discount)

leecher1337, 2022-02-15:
At least on L530, updating with a non-geniune battery installed worked without any issues, the updater didn't care, but I guess this is not comparable to your situation, as L530 incorporates EC code in SPI flash BIOS chip and not - like on your machine - in the EC internal memory that needs to get updated in a seperate step.

Xanth0k1d, 2022-03-03:
To update EC on x230t, or possibly x230 as well:
1. Battery: working genuine lenovo battery/3rd party battery(EC was flashed even if battery error occurs before boot)
2. Charger needs to be plugged in.

That is all.

closed 2022-03-03 by Xanth0k1d