hamishcoleman/thinkpad-ec #210: Collection of Last Known Working Firmwares

Xanth0k1d, 2022-02-09:
Anyone willing to contribute the 4mb and 8mb bin dump of their thinkpads? (Last known working version of the lenovo ones)
Will make it easier for those of us with a ch341a but no battery/forgot to back up the original EC/BIOS.
Actually the 4mb dump is readily available via extraction of FL1.

leecher1337, 2022-02-12:
I can contribute a L530 dump, if you want. Where should I upload it?

Xanth0k1d, 2022-03-04:
Got my battery, hence commenced flash of BIOS & EC.
Will update with gcuj32 and patched gcuj32. (Using flashrom in vbox, the 8mb chip took me 16 minutes to read, and since I'm reclipping the chip for each dump so I can compare the md5, it will take me a while.)
[gcuj34us.tar.zip](https://github.com/hamishcoleman/thinkpad-ec/files/8182295/gcuj34us.tar.zip)

Xanth0k1d, 2022-03-08:
gcuj32us, and the patched one
[gcuj32us.tar.zip](https://github.com/hamishcoleman/thinkpad-ec/files/8201922/gcuj32us.tar.zip)
[patchedgcuj32us.tar.zip](https://github.com/hamishcoleman/thinkpad-ec/files/8201923/patchedgcuj32us.tar.zip)

If one is interested in 1vyrain, you will need to get the bios version 2.58, corresponding to gcuj18us.iso if I am not wrong.

closed 2022-03-08 by Xanth0k1d