hamishcoleman/thinkpad-ec #211: Wrong Drive Used for Install Messup

AngelOzz, 2022-02-10:
So I already know what was my messup, I just need guidance on how to proceed.
While writting the lines of code into the terminal, I inputed the wrong drive for the modification to be installed by accident, I installed the mod into the main hard drive, the one that has my installed windows partition, and now everytime I turn the laptop on, it indicates "missing operating system". not even the usb with my instance of ubuntu is accessible as a boot option ("error : no such device"). 
How do I go about fixing this while maintaining the best chances of recovering my data?
Or just straight up let me know if all my data is terminally gone.

AngelOzz, 2022-02-10:
> It's going to be pretty dirty if you try to fix your windows from booting or something. Hence my suggestion: Boot ubuntu and see if your windows file are there(your ubuntu usb drive should still be working, if you cant boot from it then recreate a bootable ubuntu usb and try again) If you see a bunch of mess/your windows partition doesn't even show up in ubuntu, then your data is gone. (You could take your hard drive to a recovery center and try your luck then, if your data is really that important) If your data is still there, just copy out whatever you need and reinstall windows.

Yeah the partition is gone, there is only a bunch of files that seem to be the installation files of the modification. Do you know if there is any way that I can recover the files by myself? any program recovery software that could work? I mean, I didn't have anything terribly important there, but it would still be useful to at least look at what I lost.

hamishcoleman, 2022-02-10:
@AngelOzz 
You have pretty much got one of the worst case scenarios here.

With some understanding of your disk layout, you might be able to recover the filesystem, but it is tricky and can be time consuming.

If you had formatted your disk with EFI partitioning, there is a backup partition table located at the other end of the disk that you can use, but since you are on a X220, this is not likely.

Modern Windows installs will have the first partition on the disk as a Recovery partition - probably around 500 Meg in size.  Again, the X220 is a little too old for me to guess what your layout is.

However, if you did have this recovery partition, then that is all you have overwritten (along with the partition table), so a disk recovery tool should be able to find the start of the real system partition where your files are stored.

Always remember though, the first rule of data recovery - do not try to do the recovery on your original disk, take the drive out of the laptop and make a copy of it - only ever make changes to the copy and then you can try several different methods to restore.  (this is helped with modern tools like Virtual Machines, but this gets deeper into the "requires expert knowledge" area)

Xanth0k1d, 2022-02-11:
Yeah, let me try to add in my very limited understanding:

1. dd is byte by byte copy, meaning if you did "dd if=patched.img, of=yourwindowsdisk":
You wiped off a few megabytes from the begining of your windows disk (the size of the patched.img)

2. If you had a recovery partition/MSR partition or your installation is in efi mode, technically your main windows partition aka C drive should not be touched (Any of those partitions are definitely larger than the patched.img and they should come before your C drive, hence you'd probably only damaged those partitions)

3. The question is, when you did dd, did you wipe off anything that will affact you from mounting the C drive (You wiped the mbr? you wiped the first few bytes which is required for OS to see what partitions are there)? - this is something I'm not sure

4. Did you do "dd if=xx.img of=/dev/sdx" or "dd if=xx.img of=/dev/sdxX"? If you did the second one, you probably wiped the partition, which means you indeed have to go to a data recovery lab...

Xanth0k1d, 2022-02-11:
[image]

Yup, you are pretty much toast. This is what I did to a mbr install of Windows and here's the result:
[image]

So now it can be concluded that your data is probably still there, but you will need to restore the partition table.

Xanth0k1d, 2022-02-11:
Before you do anything, do as back up as hamish had suggested.
You need to confirm if your disk is mbr or gpt. (A simple say to start is to recall if your windows is booting via efi or legacy)
If it's a mbr disk: https://superuser.com/questions/56209/vista-wiped-mbr-and-head-of-disk-how-to-recover?rq=1
If you are using EFI, I'm not sure how you can proceed, but probably copying the back up table to the start of the disk.

AngelOzz, 2022-02-21:
> [image]
> 
> Yup, you are pretty much toast. This is what I did to a mbr install of Windows and here's the result: [image]
> 
> So now it can be concluded that your data is probably still there, but you will need to restore the partition table.

Yeah my partitions basically looked like that second image you posted. I did manage to recover some stuff from the hard drive by using testdisk on the unallocated partition table. Thanks for everyone's help.

closed 2022-02-21 by AngelOzz