hamishcoleman/thinkpad-ec #216: patches not being reversible

nullquine, 2022-05-10:
hello, I would like to share my experience, I am not entirely convinced it is the intended behavior:

once you create a patch with the keyboard patch enabled, it is not reversible by another one which has this patch disabled.

It makes sense but some might find it troublesome as did I, I mistakenly applied a modification with the keyboard patch enabled and now I cannot revert it.

maybe add this highlight to the readme

hamishcoleman, 2022-05-10:
It is reversible with a build that has the keyboard patch disabled.  Perhaps you experienced an issue with the second flash update.

If you can share the build report, I can at least check that things built as expected for you.

hamishcoleman, 2022-05-10:
Well, your built FL2 hash matches the one I get.

We are directly calling the flashing tool - bypassing the usual version number checks.

There are a number of people who have reported that the final flash step sometimes doesnt happen - the readme has a list of the known conditions, so I can only suggest that you have go through and try all that

nullquine, 2022-05-10:
Hereby I want to sincerely apologize, it was an user error from me:

I use a docking station, so for charging I just dock my laptop. I just rebooted my laptop with the charger plugged into it directly and it flashed the EC FW. I missed this crucial detail and was also under the impression the DOS flasher itself also checks the power source, which is not the case.

Thank you for your patience and your help, now it works perfectly.

closed 2022-05-10 by nullquine

hamishcoleman, 2022-05-10:
You are welcome!  Glad it worked in the end.

We have ended up with a lot of folklore for making the flash complete, as we are essentially reverse engineering the lenovo process.

nullquine, 2022-05-10:
Thank you for your hard work, it is insane :)