hamishcoleman/thinkpad-ec #217: Will it work for x230i?

bitsn1per, 2022-05-28:
Thank you for sharing your work and even making video for it! I'm so much fascinated with it. But, for a pitty I have only x230i version now. I would like to know is there any difference in firmware at all?

I've been searching on the web and found no solution for my model. Do you know if I can use this firmware for x230i or not? Thank you in advance!

hamishcoleman, 2022-05-28:
According to the Lenovo BIOS download webpage (https://pcsupport.lenovo.com/us/en/downloads/ds029187) the x230i is functionally identical to the x230, so you can use the x230 patches in this repo.

bitsn1per, 2022-05-28:
Thank you so much for your response! Keep the good job going!

msoczewka, 2025-09-18:
@erdosxx please describe the process - what you're running, what happens. Does the firmware update run? Is the temporary image written? Does the actual update run? Any errors, messages?

Whatever you're doing I suspect the EC memory chip just does not get the new code and that's it. Maybe some UEFI settings, maybe the update prevention setting, checksum, too new initial version?

As written above, x230i == x230. AFAIK the firmware is the same one, so image should be the same.

erdosxx, 2025-09-19:
According to your suggestion, following is my configuration.

A. Original firmware
When restore original firmware, (See below figure)
1. BIOS version: 2.51
2. EC version: 1.10
The problem in this firmware is that I cannot upgrade to 2.75 with
Lenovo exe file because I cannot charge my battery. 
As you know, when I run installer exe file, it checks AC power and battery together but my laptop supports AC mode without battery or battery alone. (This issue was resolved by installing coreboot(Skull) but 
to install your thinkpad-ec, I need to restore to stock bios.)
[image]

B. Update to 2.75 with img extracted from ISO file.
To resolve above issue, I extracted FL1 file from Lenovo ISO by using geteltorito tool.
Using flashrom tool I copied it to my top rom.
1. BIOS version: 2.75
2. EC version: 1.10
As you can see in the following figure, only BIOS is upgraded and
EC is still 1.10. 
[image]

I think that to install thinkpad-ec, I need to upgrade EC to 1.14.
Are there any solution to upgrade EC version without using official
Lenovo exe file?

msoczewka, 2025-09-19:
I believe you might have just answered your initial question. To patch the controller's firmware properly I'd first full up(down)date to 2.74 or 2.75 BIOS version and then apply the EC patch.

What you need sounds doable without flashing the memory chip directly, but I don't have the answer at hand. When I find a moment I'll take a look at my setup and try to figure it out for you.

As a side note I think this community needs a firmware updating guide to help solve individual problems. Maybe I'll write one.