hamishcoleman/thinkpad-ec #221: Help Lenovo v580c

deeptenk2, 2022-07-17:
Hello, my english is not very good, sorry.
I need help, Lenovo V580c 20160, i need to disable the authentic battery validation check permanently.
I once managed to do something similar with the t430, but I can't figure out how to build an iso image for a usb stick for the v580c.
Can you help me with this? 
https://ibb.co/3Sw4xzD

deeptenk2, 2022-07-19:
missing 0 byte files in repo

leecher1337, 2022-07-19:
Notes to the maintainer:
To be more specific: The 0 byte files in merged repository are there on purpose:

https://github.com/leecher1337/thinkpad-ec/tree/Lx30/b580.H1ET73WW.img.d
https://github.com/leecher1337/thinkpad-ec/tree/Lx30/b590.H9ET92WW.img.d
https://github.com/leecher1337/thinkpad-ec/tree/Lx30/e330.H3EC35WW.img.d
https://github.com/leecher1337/thinkpad-ec/tree/Lx30/l430.G3HT40WW.img.d

Otherwise, the script bails out with an error not finding the appropriate patches,
when not disabling Keyboard-Patch.
Maybe readd the 0byte files or modify Makefile to ignore missing files, i.e. 
if user doesn't disable keyboard patches? (See: https://youtu.be/us6K5udvJbI )

Additionally, update documentation so that it ensures that innoextract
gets installed (add it to apt-get install instructions or make build-deps)

hamishcoleman, 2022-07-19:
@leecher1337 I removed the zero byte files because you cannot patch something that you cannot patch - if you try to patch the keyboard then it will fail because it cannot patch the keyboard.

You are right that the docs should have mentioned to install the innoextract

hamishcoleman, 2022-07-19:
@deeptenk2 , it looks like you did not run any configure steps.  At step 6, you should:
```
make patch_enable_battery patch_disable_keyboard clean
```

deeptenk2, 2022-07-19:
Yes thank you. I already figured it out. I'm a Windows sysadmin and don't know much about Linux.p.s. everything worked out and went well. 20.07.2022, 02:25, "Hamish Coleman" ***@***.******@***.*** , it looks like you did not run any configure steps. At step 6, you should:make patch_enable_battery patch_disable_keyboard clean

hamishcoleman, 2022-07-19:
Good to hear!  Thanks for letting me know.

closed 2022-07-21 by deeptenk2